ArduPilot/ardupilot #7633: NKF4.SS and EKF_STATUS_REPORT.flags out-of-sync

peterbarker, 2018-01-31:
#### Issue details
We have traditionally kept dataflash's NKF4.SS and EKF_STATUS_REPORT.flags in-sync, so if you want to decode these flags you can look anywhere these things are found - in ardupilotmega.h or in the EKF libraries.

Unfortunately we seem to have gotten out-of-sync; the gps-glitch flag is at a different offset:

https://github.com/ardupilot/ardupilot/blob/master/libraries/AP_NavEKF3/AP_NavEKF3_Outputs.cpp#L581

vs

https://github.com/ardupilot/ardupilot/blob/master/libraries/AP_NavEKF/AP_Nav_Common.h#L35

(the filter status is directly copied into `NKF4.SS`)

AFAICS we have three options:
 - keep things as they are.  If you want to decode the meaning of the mavlink message you look at `ardupilotmega.h`.  If you want to decode `NKF4.SS` you look in `AP_NavEKF?_Outputs.cpp`.
 - change the way we construct the mavlink message (change that `<<15` to `<<14` (but using an mavlink enumeration value))
 - change the way we construct the dataflash logs (insert a dummy bit at bit-offset 14, pushing the glitching flag to 15).  We can come back and re-use bit 14 for something else later.

By default the first option there will be the case.

@priseborough do you have any strong opinion here?

#### Version
master
#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
n/a

#### Hardware type
all

#### Logs
n/a

priseborough, 2018-01-31:
I'm happy to leave things where they are within the main code and the bit used by EKF3 to 14

IamPete1, 2021-09-12:
It sounds like the plan is to keep it how it is. Now we have log documentation we could document the SS enum there?

peterbarker, 2024-05-24:
Closing this as it's had NeedAnswer a very long time.

closed 2024-05-24 by peterbarker