ArduPilot/ardupilot #7634: No way to disable logs

night-ghost, 2018-01-31:
If I try to set LOG_BACKEND_TYPE to 0 then I got

PreArm: Logging failed

[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2018-01-31:
This is not really the place to raise support issues.  Support issues should go into discuss and then when they are confirmed to be bugs we raise them here.  Sorry to be strict but if we're not then this just turns into yet another support forum.

Try setting LOG_BITMASK to zero.

night-ghost, 2018-01-31:
Log bitmask is not the best way because requires to consult with docs to turn it back on. Would not it be more logical to allow the LOG_BACKEND_TYPE to be used to turn logs off?

rmackay9, 2018-01-31:
That's a reasonable suggestion but then we should change the title because it's incorrect to say, "No way to disable logs".

peterbarker, 2018-01-31:
On Wed, 31 Jan 2018, Randy Mackay wrote:

> Try setting LOG_BITMASK to zero.

Weeeelll... I'd really rather say, "turn the relevant bit off in 
LOG_BITMASK" :-)

peterbarker, 2018-01-31:
On Wed, 31 Jan 2018, night-ghost wrote:

> Log bitmask is not the best way because requires to consult with docs to
> turn it back on. Would not it be more logical to allow the LOG_BACKEND_TYPE

You can set the LOG_BITMASK to 0.

You can also turn the relevant bit off in the arming check.

night-ghost, 2018-01-31:
see description of LOG_BACKEND_TYPE parameter. 0 - legal value wich means "no ligging". 

The question: why this valuse cause failed pre-arm checks?

PS. I certainly understand that it is easier to suggest using roundabout ways than to repair :)

peterbarker, 2018-01-31:
On Wed, 31 Jan 2018, night-ghost wrote:

> see description of LOG_BACKEND_TYPE parameter. 0 - legal value wich means
> "no ligging".
> 
> The question: why this valuse cause failed pre-arm checks?

Because by setting the bit in the arming mask you have specified that 
logging *must be working* for arming to happen.  If you have no backends 
specified, no logging can happen.  So you will get a pre-arm failure.

night-ghost, 2018-01-31:
>Because by setting the bit in the arming mask you have specified that
logging *must be working* for arming to happen

No, this is is wrong logic that requires additional efforts from user. Human logic: logs should be if I did not turn them off explicitly. And LOG_BACKEND_TYPE is the best point to turn it off.

Pedals2Paddles, 2018-01-31:
I agree with @peterbarker . This is user-error conflicting parameters as currently implemented.  The arming check is being told to verify logging works, and the user has it set to do so with the `arming_check` parameter. But elsewhere, you have turned off logging. So it is rightfully telling you something is not as desired.  You can disable logging checks with a value of 7166 in the `arming_check` parameter.

So this isn't a bug or something that needs "repair" IMO. It is doing exactly what it is supposed to do.  However there certainly may be some merit to programatically skipping the arming check if the `LOG_BACKEND_TYPE` is set for zero. **PRs welcome to change it**, not a bad idea!

> PS. I certainly understand that it is easier to suggest using roundabout ways than to repair :)

You might get more desirable responses, or more people willing to help you, if you didn't insult them with sarcastic comments while asking for their help.

night-ghost, 2018-01-31:
Oh, one more amateur to prove "we are doing well, you do not need it."

I already fixed it in my fork, so just wanted to help. Do not want to - do not.

auturgy, 2018-01-31:
I think NG has made a valid point, one that from a user perspective is far broader than just logging. The pre-arm bitmask isn’t hooked into the associated Params.
The same applies to joystick control, no-gps, etc.
I don’t think this is a ‘bug’, but is a valid feature request.

Regards,

James

peterbarker, 2018-01-31:
On Wed, 31 Jan 2018, Matt wrote:

> exactly what it is supposed to do. However there certainly may be some merit
> to programatically skipping the arming check if the LOG_BACKEND_TYPE is set
> for zero. PRs welcome to change it, not a bad idea!

I feel responsible when people go out and fly their vehicles to actually 
get work done, but come home to find they have no logs.  I would prefer to 
err on the side of caution here - if the user has specified that logging 
must be working for the vehicle to arm then that should be enforced.  It's 
a one-stop shop.

I'd be OK to be overridden here.  I'll just tell everyone Matt made me do 
it :-P

Pedals2Paddles, 2018-01-31:
Lol. I think there's also lots of merit to these critical systems with arming checks working exactly how they're working now. So both ways have their pros and cons. IMO, the pros of the way it's done now outweigh the cons and it should stay. But to that end, it probably should be mentioned in the documentation of those parameters to avoid user confusion.

rmackay9, 2018-01-31:
@night-ghost, please be careful not to get personal.  I've edited your response above.

We are discussing an issue.  There's no need for comments about someone's character just because they have a different point of view.  Please take this as a warning.

night-ghost, 2018-01-31:
>please be careful not to get personal

@rmackay9, that "person" yesterday openly called unfamiliar to him people the idiots, and instead of apologies he began to mock. I don't want to deal with, sorry.

closed 2018-01-31 by night-ghost

fnoop, 2019-01-31:
We've been here before!
https://github.com/ArduPilot/ardupilot/commit/268d6523e9c76e0e316bad0d2472c987a431d91d
https://github.com/ArduPilot/ardupilot/issues/7308

peterbarker, 2019-01-31:
@fnoop Yes, previously I said "no" to various related options - and I'm pretty sure to the exact thing that was merged.  It's not a small change to make to the software, but we discussed it on the devcall and there was general agreement.

fnoop, 2019-01-31:
PR #10265 is a great step forward for user-friendliness.  It would also be nice to resolve #7308, so you could either:
 - Run the logging pre-arm check to ensure that *at least one* logging is working, or
 - Add a LOG_BACKEND_TYPE as @tridge suggested that allows an optional backend type that doesn't affect the arming checks, or
 - a LOG_BACKEND_HEALTHY that @OXINARF suggested that allows a bitmask of log backends to check, or
 - Fix parameter change so a firmware reboot isn't necessary

mrnams, 2021-09-05:
I had same problem, My Flight Controller Pixhawk 2.4.7 having Copter 4.0.7 was giving pre arm error “ Logging failed : Logging pre-armed was enabled but failed to write to the log.”
What I tried
1.	Formatted memory card- Not worked
2.	Replaced memory card with new one- Not worked
3.	Installed fresh Firmware Copter 4.0.7 and calibrated all- Not worked
Finally I got below solution which worked for me
Formatted memory card with File System “FAT32” -Worked