ArduPilot/ardupilot #7636: RCInput_UDP feature request - Navio2 hardware

drobban, 2018-01-31:
#### Issue details
Needed Joystick via network functionality. So I read the ardupilot source code to find out where to incorporate this functionality. 
As I was doing this, I found that someone else before me had done this for the Bebop hardware.
So I changed some lines to add support for this in the Navio2 hardware as well and is now wondering if this is something that the ardupilot would accept an pull request on?
If so, I just have to make it "commit-able" before I can submit an request.

RCInput_UDP for Navio2

#### Version
Arducopter-3.5

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Hardware type
Navio2

auturgy, 2018-07-30:
I’m not sure that UDP is the best protocol for control (as there’s no guarantee of delivery), but it probably isn’t really worse than many others. Pull requests are always welcome!

drobban, 2018-09-05:
Sorry for the late reply, must of missed the notification.
I have a test version I run my own hardware.
The version I run with isn't production quality, but that is something I intended to change in the event of someone giving a go on this issue.

Been a while since I messed with the code, but if I remember correctly there is configuration options for radio and such and that I planned to extend the configs with UDP options.

Agree with @peterbarker regarding UDP.

peterbarker, 2021-07-30:
@drobban are you still interested in chasing this?

IamPete1, 2021-08-15:
No response so i'm going to close, I think we answered the question anyway, "yes a PR for that would be Great"

closed 2021-08-15 by IamPete1