ArduPilot/ardupilot #7640: wind speed and direction

JeyPi, 2018-02-01:
The newest arduplane software is imterpretting the wind speed x direction wrongly
#### Version
_What version was the issue encountered with_
Arduplane 3.8.4
Mission planner  1.3.5.2

Platform
Plane

Airframe type
plane Vtail

Hardware type
Pixhawk

Logs
[Tlog.zip](https://github.com/ArduPilot/ardupilot/files/1685179/Tlog.zip)

lthall, 2018-02-01:
It would be helpful if you would elaborate on this further.

JeyPi, 2018-02-02:
No problem I can do that as long as it is about testing and flying

Raviu8, 2018-02-05:
I can confirm on this, I thought it was my fault until I saw this report.
I've added a ::printf() to test with sitl and I get strange results.

When I set the sim_wind_* to something specially the direction, I don't get near values with `degrees(atan2f(-wind.y, -wind.x)) ` . direction is not as expected even after doing a lot of loiter turns.

magicrub, 2018-02-05:
@priseborough have you heard anything?

WickedShell, 2018-02-05:
SITL creates the correct wind estimates for me.

@JeyPi a dataflash log would probably help. I do agree that it looks a bit surprising on start, but without a DF log I don't think it will make it to much further. I have seen the wind estimate blow up real world shortly after takeoff due to compass upsets among other things.

@Raviu8 An important note on SITL is that large step inputs will confuse the wind estimator and cause it to blow up. Once it's blown up badly enough it won't ever reconverge, so adjusting the wind parameters in SITL needs to be done relatively slowly. (Ideally we would probably limit how fast we let it change in the sim model, but no one has implemented that yet.

ShikOfTheRa, 2018-04-12:
Can anyone confirm if the wind direction reported in Mavlink is relative to the heading of the plane or North?

WickedShell, 2018-04-12:
North, not plane heading.

ShikOfTheRa, 2018-04-13:
Thanks for response.
 
That's what I read, but when plane turns, wind arrow on OSD turns which I wouldn't expect if ground ref north used.

WickedShell, 2018-04-14:
I don't use an OSD so I'm not fully sure but the OSD may be attempting to
give you wind relative to your current heading instead. I can only attest
to the data we aren't out of MAVLink. (At low wind speeds I do see a lot of
heading changes on wind estimate when turning)

magicrub, 2018-04-15:
The wind estimate improves when turning so that's probably the change you're seeing

ShikOfTheRa, 2018-04-16:
Thank you for clarifying. It helps and supports what I am seeing.

So now I need to figure out why arrow is 180 out from wind direction, but it is some result on different OSD firmware. 

One other - does the wind heading value indicate direction the wind is coming from or going to?

I have read both, but as of today I now think that it indicates the direction wind is coming from, but Mission planner indicates direction in which wind is going to. OSD's are misreporting.

auturgy, 2018-04-16:
Probably worth pointing out that @ShikOfTheRa is the lead for MWOSD, which is a popular minimOSD firmware.

ShikOfTheRa, 2018-04-16:
yeah that's me! - but I think MWOSD is a bit of an unknown in Ardupilot community however its starting to grow with over 100 new users last week:

[image]

Any Ardupilot devs wish to try it FOC just ask...

Unfortunately I don't have a plane with ardupilot on it so I rely purely on tester feedback. Copter only with pixhawk. I'm more of a raw control flyer with planes - but will put it in one soon.

WickedShell, 2018-04-16:
Ardupilot is reporting the direction the wind is coming from, it's upto the MAVLink receiver to choose how to render the direction after that. https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/ardupilotmega.xml#L1173

ShikOfTheRa, 2018-04-17:
@ WickedShell - Perfect. Your answers make sense and match what I thought. It helps a lot and I learned something. 

I think for a pilot flying using an OSD an arrow indicator pointing in the direction in which the wind is actually heading is more intuitive to show effect on aircraft.. 

I was not confident to report different wind direction to other OSD's that were established for many years when I have not much experience with mavlink.

Thanks all!

WickedShell, 2018-04-26:
Closing as this seems to be a support issue, particularly without a DF log it's hard to investigate the wind estimate, and the direction reporting part of this is a support thing as well.

closed 2018-04-26 by WickedShell