ArduPilot/ardupilot #7643: EKF3:  wrong coviriance index?

lesliefan1, 2018-02-02:
#### Issue details
_Please describe the problem, or desired feature_
void NavEKF3_core::correctEkfOriginHeight()
{
    // Estimate the WGS-84 height of the EKF's origin using a Bayes filter

    // calculate the variance of our a-priori estimate of the ekf origin height
    float deltaTime = constrain_float(0.001f * (imuDataDelayed.time_ms - lastOriginHgtTime_ms), 0.0f, 1.0f);
    if (activeHgtSource == HGT_SOURCE_BARO) {
        // Use the baro drift rate
        const float baroDriftRate = 0.05f;
        ekfOriginHgtVar += sq(baroDriftRate * deltaTime);
    } else if (activeHgtSource == HGT_SOURCE_RNG) {
        // use the worse case expected terrain gradient and vehicle horizontal speed
        const float maxTerrGrad = 0.25f;
        ekfOriginHgtVar += sq(maxTerrGrad * norm(stateStruct.velocity.x , stateStruct.velocity.y) * deltaTime);
    } else {
        // by definition our height source is absolute so cannot run this filter
        return;
    }
    lastOriginHgtTime_ms = imuDataDelayed.time_ms;

    // calculate the observation variance assuming EKF error relative to datum is independent of GPS observation error
    // when not using GPS as height source
    float originHgtObsVar = sq(gpsHgtAccuracy) + P[8][8];

- I think the last line should use P[9][9], which is the height variance in EKF3 
    float originHgtObsVar = sq(gpsHgtAccuracy) + P[9[9];

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2018-02-02:
txs for the report.  I've ping'd @priseborough so perhaps he can answer.

priseborough, 2018-02-20:
https://github.com/ArduPilot/ardupilot/pull/7745

rmackay9, 2018-02-21:
Nice find @lesliefan1 and thanks for the fix @priseborough!

closed 2018-02-21 by rmackay9

auturgy, 2018-02-21:
I love this type of pickup. A great example of how open source code enables people to critique us and makes for more robust software. 

Regards,

James