ArduPilot/ardupilot #7646: Rover: report MAV_TYPE as Boat when FRAME_CLASS =2

rmackay9, 2018-02-02:
As we get more serious about supporting boats we should make boats appear as boats on the ground station map.  To allow the ground stations to know it is a boat we should report the vehicle's MAV_TYPE (held in the heartbeat message i think) as MAV_TYPE_SURFACE_BOAT if FRAME_CLASS = 2 (aka "Boat").

starlord1311, 2018-02-12:
can i take up this issue? @rmackay9

rmackay9, 2018-02-12:
@starlord1311, yes, please!

starlord1311, 2018-02-13:
can you help me with what i am supposed to do?

khancyr, 2018-02-13:
Hello,

You need to correct the vehicle type reported first. Currently it is always MAV_TYPE_GROUND_ROVER. There is at least two place to change : see  https://github.com/ArduPilot/ardupilot/search?utf8=%E2%9C%93&q=MAV_TYPE_GROUND_ROVER&type= for example. A simple if case should be enough. 
On second time, you need to correct the GCS (Mission Planner and mavproxy). Well this can be uneasy, so opening an issue on mission planner explaining what the change should, is already a good start. On mavproxy, you need to find the function in charge of loading the vehicle image and change it to a new boat image !

starlord1311, 2018-02-13:
So i just need to add MAV_TYPE_SURFACE_BOAT next to MAV_TYPE_GROUND_ROVER wherever it is used? @khancyr

rmackay9, 2018-02-13:
@starlord1311, I think that's probably all that's required.  I've raised an issue for Mission Planner so hopefully @mee1 can add the icon for boats somehow.  Maybe someone will need to make a little icon for him.  That's beyond the scope of this vehicle related change though.

rmackay9, 2018-03-22:
fixed in master.

closed 2018-03-22 by rmackay9