ArduPilot/ardupilot #7647: Strange behavior of uart alternate function

liang-tang, 2018-02-02:
#### Issue details
  There is strange behavior of uart of our custom board, dms_init function could change ALT function of GPIOA0 from ALT8(uart4_tx) to ALT0, details as below:
[image]
[image]
[image]
[image]
[image]

#### Version
3.8.4

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
custom board

#### Logs
_Please provide a link to any relevant logs that show the issue_

liang-tang, 2018-02-02:
But when i set GPIO_USART1_TX to GPIO_USART1_TX_2, everything is ok, PA0 alternate function won't be changed. How could it be? I can't understand why.

liang-tang, 2018-02-02:
@tridge

closed 2018-06-22 by liang-tang

magicrub, 2018-06-22:
Why did you close? Was this resolved?

liang-tang, 2018-06-23:
Ignore this issue, because we will switch to ChibiOS platform.