ArduPilot/ardupilot #7649: Allow overriding mission altitude and yaw in AUTO

khancyr, 2018-02-02:
#### Issue details
I got some report from my company pilot that they prefer Mikro copter in Auto mission as they can temporary override the mission altitude and yaw. This allow them to avoid some obstacle or have a better view on the ROI. 

This could be added as an option on switch : with an action on throttle stick doing some climb/fall change, and same for yaw like it is done on POS_HOLD. You control a altitude target and not directly the throttle.
The option could be slip in two : keep the override only for the current travel to the next waypoint, or keep it until the end of the mission.

#### Platform
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

Naterater, 2018-02-02:
Plane somewhat has this feature, but it's not controlled by a channel. The parameter is called ALT_OFFSET. 

I like this idea as it allows adjustment on the fly and can make semi-automated flight a lot easier, especially if terrain data is bad in a location.

rmackay9, 2018-02-03:
Tridge's skyviper patch has this for RTL but I don't think it has it for Auto.

rmackay9, 2022-05-26:
I'm going to close this issue because it's a duplicate of https://github.com/ArduPilot/ardupilot/issues/11251

Now, normally we close the newer one but the other one has more comments so I'll keep it instead.

closed 2022-05-26 by rmackay9