ArduPilot/ardupilot #7654: Feature Request - ArduCopter

RowanWatkins, 2018-02-03:
#### Issue details

The ability to add a Airspeed Indicator to a Traditional Helicopter (or Other) and display the airspeed data on Mission Planner.

I understand that eventually airspeed may be added to the Copter code for more advanced navigation features, but in the interim if airspeed can be displayed on the ground station it will allow the user to modify the target speed values on complex missions to keep the aircraft within its safe operating speed range.

#### Version
All

#### Platform
[  ] Copter

#### Airframe type
All

lthall, 2018-02-03:
We may be able to prevent over speed in the first instance and in the second instance add the ability to maintain airspeed.

Pedals2Paddles, 2018-02-03:
Updated the title and labels.

geofrancis, 2018-02-04:
Wouldn't you need to have the sensor shielded from the down draft for it to get a stable reading

rmackay9, 2018-02-05:
Adding simple reporting of airspeed in multicopters is not too difficult.  I can't take this on immediately but if another dev wants to, here is some advice:

- include the Airspeed library by adding "include <AP_Airspeed/AP_Airspeed.h>" to Copter.h
- instantiate an airspeed object by adding "AP_Airspeed airspeed" to Copter.h
- add a "read_airspeed" function to sensors.cpp and add to the scheduler in Copter.cpp (have a look at what plane does)
- modify Copter::send_vfr_hud() so that it sends the airspeed sensor's speed if it's enabled.  Currently it seems to send the gps speed if the airspeed field which is a bit weird.  I think if we don't have the airspeed sensor we should just send the ground speed in that field according to what the ahrs/ekf thinks it is.

bnsgeyer, 2018-02-18:
I did this in 3.3.3 but never PR'd to master.  I will be looking into doing this again for the latest version once I get my compound helicopter built and ready to fly.  Can't promise that I'll get to this soon but I will put it on my list and I suspect I will probably have it implemented come summer.  Unless someone beats me to it. ;-)

closed 2022-01-19 by tridge