ArduPilot/ardupilot #7658: Copter, Rover: add support for DO_SET_ROI_LOCATION

rmackay9, 2018-02-05:
We should add support for the new ROI messages to Copter and Rover that can be found here: http://mavlink.org/messages/common

- DO_SET_ROI_NONE - cancels the ROI
- DO_SET_ROI_WPNEXT_OFFSET
- DO_SET_ROI_LOCATION

drprofpatrick, 2019-09-04:
Has this been solved yet? This would be very useful for work.

rmackay9, 2019-09-04:
@drprofpatrick, no i'm afraid not.  the regular DO_SET_ROI works of course though so I think it's possible to use this existing command to do the same thing.  It just requires a little more work creating the mission because the user will need to input the lat/lon.

rmackay9, 2019-11-11:
We've got partial support this in Copter/Rover-4.0 because we support MAV_CMD_DO_SET_ROI_LOCATION.  We still don't support these variations:

- MAV_CMD_DO_SET_ROI_WPNEXT_OFFSET
- MAV_CMD_DO_SET_ROI_NONE

billc303, 2021-04-06:
Any update as to when QGC structure scan might be implemented in Ardupilot?

rmackay9, 2023-06-08:
I think I'll close this issue and add a link to the PR to the gimbal enhancement list.

closed 2023-06-08 by rmackay9