ArduPilot/ardupilot #7661: Rover: boats could weathervane into to hold position at waypoints

rmackay9, 2018-02-05:
After this PR goes in (https://github.com/ArduPilot/ardupilot/pull/7660) boats will "loiter" around waypoints if there is a delay on the waypoint (or we're in Guided mode and have reached the target).

It would be nice if we could somehow detect the water current (and airspeed?) and use it to our advantage when loitering.  The boat could likely stay almost completely still if it pointing into the current and powered the motors just enough to counteract the current.

ncx94, 2018-02-07:
That would be an awesome feature! How do you expect to do that? Detecting how the boat's gps coordinates drift and assuming that's caused by the water current?

rmackay9, 2018-02-08:
@ncx94, there's some thoughts on implementation [in this thread](https://discuss.ardupilot.org/t/actively-loitering-issue-with-rover-3-2).  It isn't clear yet how we should do it though.

ncx94, 2018-02-08:
@rmackay9  Great thanks :)
Very interesting discussions! I've added a few thoughts on the discuss thread