ArduPilot/ardupilot #7662: Binary data on nsh console on pixhawk

Raviu8, 2018-02-05:
#### Issue details
I've been trying to use nsh console on pixhawk2 as described in the ardupilot/docs but I always get a lot of rubbish binary data on the console with high rate which make it impossible to use.
So to debug the problem I've started the pixhawk2 without sdcard connected using the following command;
`mavproxy.py --setup --master /dev/serial/by-id/usb-3D_Robotics_PX4_FMU_v2.x_0-if00`
Then executed `sh /etc/init.d/rc.APM` to see what is happening and after few seconds I started getting binary data on the console.

#### Version
I've tried many version over the past months but now I've compiled the latest source code as of writing the issue from github using ./waf on linux

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[.] Plane
[  ] Rover
[  ] Submarine

Actually I didn't try others. I always tried with arduplane.

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
PixHawk2.1

#### Logs
Attached is the terminal output
[terminal.output.txt](https://github.com/ArduPilot/ardupilot/files/1696199/terminal.output.txt)

rmackay9, 2018-02-06:
What you're seeing there is the mavlink packets normally used to communicate with the ground station.

There's some information this page about how to use the NSH shell over mavlink using the mission planner.  http://ardupilot.org/dev/docs/interfacing-with-pixhawk-using-the-nsh.html

Sorry, I don't use the NSH much myself and I'm unsure how to connect to it using mavproxy.

Raviu8, 2018-02-06:
@rmackay9 May I ask, what do you use for debugging and getting prints?

The method I use was described here http://ardupilot.org/dev/docs/learning-ardupilot-the-example-sketches.html

Thanks for the link, the mavproxy load module method worked.

rmackay9, 2018-02-06:
@Raviu8,
I use ::printf() in the code and that appears on the console in when I connect a [debug cable](http://ardupilot.org/dev/docs/interfacing-with-pixhawk-using-the-nsh.html) connected to a Pixhawk's Serial 5 port.  No mavlink messages come out on that port by default.  Although I have many Pixhawk2/cubes I actually don't know where Serial5 is on a Pixhawk2/Cube.

The other thing I do is use [SITL](http://ardupilot.org/dev/docs/sitl-simulator-software-in-the-loop.html) running on an Ubuntu VM or an actual Ubuntu machine.  ::printfs in SITL appear in the black console screen.

Raviu8, 2018-02-06:
@rmackay9 Thanks  a lot

closed 2018-02-06 by Raviu8