ArduPilot/ardupilot #7663: ARMING_CHECK is not enforced with disabled ARMING_REQUIRE

Raviu8, 2018-02-05:
#### Issue details
According to the description of ARMING_REQUIRE parameter:

>  Also note, even with this parameter at 0, if ARMING_CHECK parameter is not also zero the plane may fail to arm throttle at boot due to a pre-arm check failure. This parameter is relevant for ArduPlane only.

But actually this is not the case, I've tried with ARMING_CHECK = 1 and other values and while setting ARMING_REQUIRE to 0 and it always arm even if bad GPS health, bad gyro, bad logging and others.

The expected behavior when ARMING_REQUIRE is set to 0 and ARMING_CHECK has checks enabled is to wait until any check errors are cleared before actually arming. Like to wait for GPS fix.

#### Version
ArduPlane 3.8.x

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[.] Plane
[  ] Rover
[  ] Submarine

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
pixhawk

#### Logs
I can provide logs if required. But I think it's clear enough.

davidbuzz, 2021-07-21:
very old issue with no progress. propose closing

IamPete1, 2021-08-15:
I actually not sure sure this would be a good idea anyway. Currently you it will try and arm only at boot. With the proposed change the vehicle would just arm itself at some point after boot, it could wait 5 mins for GPS lock and then suddenly arm and take off if you have auto configured for that. I think that is just asking for trouble. At least arming at boot your expecting it.

closed 2021-08-15 by IamPete1