ArduPilot/ardupilot #7666: Add ability to trigger camera via RC

Raviu8, 2018-02-06:
I want to be able to trigger the camera using a switch on my remote control, along with the ability to do this with GCS or auto mission.
When the switch is turned on, it should do the same action as when you do "Trigger camera Now" action from MissionPlanner.
Switching off should have no effect.

Pedals2Paddles, 2018-02-06:
This is already implimented.  See the CH7_OPT through CH12_OPT parameters. http://ardupilot.org/copter/docs/parameters.html#ch7-opt-channel-7-option.  Fully described in the wiki. http://ardupilot.org/copter/docs/common-camera-shutter-with-servo.html.

Additional support and help can be found in the discuss forums: https://discuss.ardupilot.org/

geofrancis, 2018-02-06:
none of that works with plane.

Pedals2Paddles, 2018-02-06:
Ah, well in that case...

Raviu8, 2018-02-06:
Actually I don't fly copters. This is no such thing in plane

DawoodTaii, 2018-02-22:
There is an old issue #1158 related to this.
Why not just allow the same RC/Servo mapping as per other control surface. So when servo8 is selected for camera trigger, RC8 should trigger camera - once - when above 1750 ?

amilcarlucas, 2018-02-22:
You only need to add AP_CAMERA support to plane.

robustini, 2018-02-26:
Related issue #7071.

commit 901664ce (2018-09-05): Plane: add support for RCn_OPTION (servorelay,camera)

Closes #7071
Closes #7666
closed 2018-09-05 by peterbarker