ArduPilot/ardupilot #7670: bug in AP_FlashStorage

night-ghost, 2018-02-07:
(just to not forget)

See line 118. Powering on and contacts bounce, so reset occures here or bit earlier, in time of write_all(). Now we had filled out reserved space without the confidence that all data from FULL_SECTOR are stored in a IN_USE sector. So on 2nd boot write_all() will get out of space, will return false and all data will be lost.

night-ghost, 2018-02-08:
To fix it, we need a new sector state - ACCEPT_DATA, and set it before data transfer. Then if at initialization we got sector in this state, we can just to resume the transfer from the very beginning of reserved space.

peterbarker, 2018-02-20:
@lvale @night-ghost appears to have created this as a reminder to himself to fix something.

night-ghost, 2018-02-22:
This bug was found when @tridge asked me why my HAL don't uses AP_FlashStorage. That PR is now closed so this issue to not forget about bug. 

reliable EEPROM emulation is very hard to do, and some bugs are only shown itself in the case of "resets in bad time," that's one of these bugs. However, in this case the probability of the bug is quite high, since the time window is rather big because reset should occur during the page transfer.

tridge, 2019-01-21:
@night-ghost I've included a fix for this in 
https://github.com/ArduPilot/ardupilot/pull/10292
I'd appreciate it if you could check the fix. The bug this fixes is in this situation:
 - first sector is in AVAILABLE state
 - 2nd sector is in FULL state
 - on bootup we choose 2nd sector as good copy
 - we started a write_all() to copy 2nd sector over first
 - that included the sector header, which means we were making the 1st sector FULL (that is the bug)
 - if we reboot part way through the write_all() then we would choose first sector as primary, but it would only be half-done

The fix is to not write the header until we've written all the rest, so we know that we can't end up with a partial sector
Sound ok?

closed 2019-01-21 by tridge

sh83, 2019-03-27:
@tridge I think you misunderstand the problem. write_all do not write sector header. So the fix do not fix anything ...

PR #10292 (2019-01-21): SITL: enable flash storage option
This allows AP_FlashStorage to be used as a SITL backend, to enable easier testing

It also fixes #7670
commit 91e5f562 (2019-01-21): AP_FlashStorage: fixed a race condition

fixes #7670

thanks to night-ghost for the bug report