ArduPilot/ardupilot #7682: Master of today doesn't build for px4-v2

olliw42, 2018-02-09:
just to mention it
it appears that the master of today, 9.Feb., doesn't build for -v2, it does for -v4, and there wasn't any issue with master of two days ago, 7.Feb.
for -v2 I get a flash overflow error
```
LINK:    /c/Users/Olli/Documents/GitHub/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build/firmware.elf
c:\pixhawk_toolchain\toolchain\bin\arm-none-eabi-ld.exe: c:/Users/Olli/Documents/GitHub/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build/firmware.elf section `.text' will not fit in region `flash'
c:\pixhawk_toolchain\toolchain\bin\arm-none-eabi-ld.exe: region `flash' overflowed by 6476 bytes
make[2]: *** [/c/Users/Olli/Documents/GitHub/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build/firmware.elf] Error 1
make[2]: Leaving directory `/c/Users/Olli/Documents/GitHub/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build'
make[1]: *** [/c/Users/Olli/Documents/GitHub/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build/firmware.px4] Error 2
make[1]: Leaving directory `/c/Users/Olli/Documents/GitHub/ardupilot'
make: *** [px4-v2] Error 2

```

khancyr, 2018-02-09:
it build one my computer but on linux...
could you make clean and rebuild ?
What toolchain are you using
Are you using WSL ?

olliw42, 2018-02-09:
sorry, I forgot to mention this
I'm on windows and use the PX4 toolchain with Eclipse
in order to be sure it's not something I do I actually took a fresh clone of master, cleaned and all that

Pedals2Paddles, 2018-02-09:
Travis failed on the last commit from yesterday with that error on px4v2.  But appears to be working after Randy's most recent commit today.  If you fetch/pull the latest from master and try again, it may work for you.  I just restarted the travis build again to make sure.

olliw42, 2018-02-09:
thx
I do have the commit of Randy, 2aeda4851, in my branch 
(one can double-check here https://github.com/olliw42/ardupilot/commits/master)

Pedals2Paddles, 2018-02-09:
hmm.  Submodules updated?

rmackay9, 2018-02-09:
That error indicates that our build for the regular Pixhawk has gone over the 1MB limit.  If that's the case then we will probably need to remove a feature or two.

olliw42, 2018-02-09:
@Pedals2Paddles 
I use GitHub for cloning, and - so far - it always did that
I also think that if the submodules would not be updated the compile would fail earlier (remind you, this is on a fresh clone of master)
@rmackay9 
yes
it should be reproducible, I think, i.e. not only me should find that
if it happens that it just me I would be deeply sorry for raising the issue, but I can't see it

khancyr, 2018-02-09:
and we need to refactor the wiki to show people how to use waf on windows ! ;-P

Pedals2Paddles, 2018-02-09:
> That error indicates that our build for the regular Pixhawk has gone over the 1MB limit. If that's the case then we will probably need to remove a feature or two.

Which flight controllers have that 1MB limit and require v2

rmackay9, 2018-02-09:
I've just reproduced the issue as well :-(.  So the AP_Winch and AP_WheelEncoder libraries would be good one to remove.  The AP_Winch is a real edge case used for delivery that I can't imagine is going to be missed by many people.

rmackay9, 2018-02-09:
@olliw42, this isn't a real solution but for the moment, you may find that px4-v3 works on your boards.  Only the old Pixhawk1 boards with the 1MB flash limit need to use -v2.  There's a video on my youtube channel which shows how to figure out if your board has the 1MB problem.

https://www.youtube.com/watch?v=BNgClHf9nIo

Pedals2Paddles, 2018-02-09:
Are any of the cheap clones also subject to the 1MB limit? Or is it just the older revisions of the authentic 3DR Pixhawk 1?

auturgy, 2018-02-09:
There was an errata in early f427 mcu’s that meant the 2nd mb of flash wasn’t accessible, but this was rectified by ST. Any recent “Pixhawk”, or board using the same mcu should have the full 2mb accessible, regardless of manufacturer. Of course if it’s old stock, or a different mcu, no guarantees.

Regards,

James

olliw42, 2018-02-10:
thx for the suggestion, Randy
sadly, I happen to know that it has the 1MB issue, its not rev3
btw, it's a AUAVX2 of the first series

no worries, it is not a real issue for me, since I can stay with master of 2 days ago, the main purpose was to report the issue so it's known

which it is now :)

lvale, 2018-02-10:
>Are any of the cheap clones also subject to the 1MB limit? Or is it just the older revisions of the authentic 3DR Pixhawk 1?

It's irrelevant the origin of flight controller, except for the newer models like the Cube/PixRacer/others.

lvale, 2018-02-10:
Just pulled from master and built for px4-v2 and can reproduce @olliw42 issue.

And I'm not on Windows

./waf configure --board px4-v2

```
/usr/local/Cellar/gcc-arm-none-eabi/20160928/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: src/firmware/nuttx/firmware_nuttx section `.data' will not fit in region `flash'
/usr/local/Cellar/gcc-arm-none-eabi/20160928/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: region `flash' overflowed by 140 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Waf: Leaving directory `/Users/Luis/px4/ardupilot/build/px4-v2'
Build failed
 -> task in 'bin/arducopter' failed (exit status 1): 
	{task 4475236512: cmake_build_task  -> }
['/usr/local/bin/cmake', '--build', '/Users/Luis/px4/ardupilot/build/px4-v2/modules/PX4Firmware', '--target', 'build_firmware_px4fmu-v2']
Sparkle:ardupilot Luis$ 

```

muramura, 2018-02-10:
I tried to make @rmackay9 comment.
I made the winch optional.
However, I over 3K.
===
/home/murata/work/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build/firmware.elf
arm-none-eabi-ld: /home/murata/work/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build/firmware.elf section `.text' will not fit in region `flash'
arm-none-eabi-ld: region `flash' overflowed by 3460 bytes

zhaohailongwang, 2018-02-12:
I expanded the flash MEMORY 1008K =》1016K，in the ld.script. now the master can be build.But I wonder if I am doing this right

rmackay9, 2018-02-12:
This should be fixed in master now.  We removed a few new-ish features from the -v2 build including the crop sprayer, the gripper, the winch and the new "flow hold" mode.

closed 2018-02-12 by rmackay9

DawoodTaii, 2018-02-12:
Shouldn't the migration to ChibiOS solve this? As you mentioned that it generates much smaller firmware image?

olliw42, 2018-02-12:
ok, I confirm that master of today now also compiles for v2 on my end

**BUT**
it apparently does so only VERY VERY narrowly
dec is now 1087424, while before it was something with a 9, i.e., 9xxxxx
which means that in these two days things were added which consume more than 87424 bytes !!!

the problem, my fork still doesn't compile, it's short of 15k :(

I would think that many who have just added some pieces of own code will encounter this problem too; the currently available margin is very narrow. 
I'd like to encourage an investigation where these 90k are coming from, and to add more options in e.g. APM_Config.h, config.h to disable the "big" newly added features.

Anyway, to get around it I poked into the code, and in APM_Config.h, config.h I found still plenty of features I don't need and could disable. Two question please:

1: I see HAL_MINIMIZE_FEATURES and assume it is set when the v2 is compiled. In order to add my own disables properly, should I use `#if HAL_MINIMIZE_FEATURES` or `#ifdef HAL_MINIMIZE_FEATURES` ? I believe it's the first, but thought I better ask.

2: Is version.h included before config.h, i.e., could I place my disables also in version.h?

THX,
Olli

magicrub, 2018-02-12:
@dawoodtaii yes, very much so. But there will be a transition time as confidence builds as it gets deployed to millions of drones.

rmackay9, 2018-02-12:
@olliw42, thanks for the details.  It really must be related to the ChibiOS addition or support for the SkyViper.  @tridge and @bugobliterator are the experts on these two new items.

olliw42, 2018-02-12:
I went a bit through the many PRs in that period, and - without really understanding anything - to me it looks it's a lot sensor and SkyViper stuff.

rmackay9, 2018-02-12:
@olliw42, we discussed this on the dev call today and Tridge mentioned that quite a few new features went in with the SkyViper code including (but not limited to):

- updated (and more accurate) declination tables
- per-motor compass compensation
- "toy mode" which is really a transmitter with buttons

So we need to think about what to drop out but it may or may not be the new features.

kd0aij, 2018-02-13:
No problems building master (commit 9eda7) on Ubuntu 16.04 with gcc 4.9.3, about 30K bytes to spare for plane.

```
arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]

./waf configure --board px4-v2
./waf plane
...
Target         Text     Data  BSS    Total  
--------------------------------------------
bin/arduplane  1017584  2856  57208  1077648

./waf copter
...
Target               Text     Data  BSS    Total  
--------------------------------------------------
bin/arducopter       1015944  2860  57912  1076716
bin/arducopter-heli  1001640  2860  57720  1062220

```

olliw42, 2018-02-13:
thx for the update. and thx for looking at it
:)

I myself e.g am never using the motor-compass stuff, I rely on external mags  

yeah. magnetic poles are moving scaringly fast these days .. we are overdue a flip by 1/2 million years or so anyhow LOL