ArduPilot/ardupilot #7683: An issue about hard landing in Plane::crash_detection_update( )

luweikxy, 2018-02-09:
Hello, please look at this line:
https://github.com/ArduPilot/ardupilot/blob/2a81792b417ed98df79505576ceb896e8d6ef807/ArduPlane/is_flying.cpp#L221
The comments above this line, we say that should trigger hard landing event **right away**, but this lie(#L221) is to **delay** triggering the hard landing, because it use "+" not "-", i.e.
```‘C++’
crash_state.debounce_timer_ms = now_ms + CRASH_DETECTION_DELAY_MS;
```
should be like this:
```‘C++’
crash_state.debounce_timer_ms = now_ms - CRASH_DETECTION_DELAY_MS;
```
This is only what I think about, and I am not sure it is right. 
Thanks!

magicrub, 2018-02-09:
I think you're right, thanks for finding this!

magicrub, 2018-02-09:
closed via https://github.com/ArduPilot/ardupilot/commit/50a2d824c44a703e54089dec8029e8b47d6c05b1

closed 2018-02-09 by magicrub