ArduPilot/ardupilot #7706: Copter/Rover: vehicles should back away from objects

rmackay9, 2018-02-13:
As mentioned on some of our avoidance related wiki page ([SF40C](http://ardupilot.org/copter/docs/common-lightware-sf40c-objectavoidance.html), [dev avoidance page](http://ardupilot.org/dev/docs/code-overview-object-avoidance.html)), Copter and Rover will stop before hitting something but they will never back away from an object so we should add support for this.  Note that this is true both horizontally and vertically (i.e. a copter flying through a cave will not descend in order to avoid hitting the roof).

We can probably accomplish this by:

- add AVOID_REVXY_SPEED and AVOID_REVZ_SPEED parameters to AC_Avoidance/AC_Avoid library.  This params will hold the max speed that the vehicle will reverse or descend.
- modify the horizontal and vertical velocity adjustment functions in AC_Avoidance to allow the final velocity to be in the reverse direction of the input velocity (see [AC_Avoidance/AC_Avoid::adjust_velocity_z ](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_Avoidance/AC_Avoid.cpp#L118) and [AC_Avoid::adjust_velocity](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_Avoidance/AC_Avoid.cpp#L63)).  For example, for the vertical velocity adjustment (i.e. the part which stops the vehicle from hitting the altitude fence or the roof) the "adjust_velocity_z" function has a line like below after which it goes to handle the case where the vehicle has breached the altitude fence.   Currently it reduces the climb rate to zero but it could instead set it to AVOID_REVZ_SPEED.

> if (alt_diff <= 0.0f) {

oneTimePad, 2018-02-19:
@rmackay9 I was working on this issue and noticed something. 

So for  `AC_Avoid::adjust_velocity` for copter, if I am going about this in the correct way, should return the reverse velocity of what the user wants with a speed equal to AVOID_REVXY_SPEED. This should occur when the vehicle is within the fence margin or has breached the boundary. These are the two conditions in `AC_Avoid::adjust_velocity_polygon`: `if(_fence.boundary_breached...` and `if(limit_distance <= margin_cm)`.  Next time the copter's velocity controller loops around it will calculate the difference between the user's desired velocity and the current one (which is in the reverse direction if the pilot continues to want to go in the same direction as before), since it limits the change in velocity, the desired velocity might still be in the reverse direction of what the pilot wants. Which means `AC_Avoid::adjust_velocity` will change the boundary points it is looking at, and it will not longer set the speed to AVOID_REVXY_SPEED instead it will limit it as well (the `else` for `if (limit_distance <= margin_cm)` in `AC_Avoid::adjust_velocity_polygon`) . This causes the copter to go in the reverse direction, but slow down to 0 then speed back up in the reverse direction of what the pilot wants and slow down again. Once the speed hits zero, the case `if (limit_distance <= margin_cm)` becomes true again and the speed gets reset to `AVOID_REVXY_SPEED`.

I thought a way to avoid this is to make `Copter::adjust_desired_velocity_with_accel_and_fence_limits` aware that `AC_Avoid::adjust_velocity` has flipped the input velocity. This is so it doesn't attempt to compute the required change in velocity to achieve what the pilot wanted and instead passes in the actual desired velocity of pilot to `AC_Avoid::adjust_velocity`. Thus `AC_Avoid::adjust_velocity` will look at the correct boundary points for the pilot's input velocity. It does this until the pilot decides not to go towards the object the velocity input is leading the copter towards, or until the copter is out of the margin. Was wondering if this change was ok, or if there might be a better way to do this? Or is this functionality desired?

Hope my explanation makes sense!

rmackay9, 2020-12-18:
This is implemented in master now and will go out with AP 4.1 (Copter and Rover) so closing

closed 2020-12-18 by rmackay9