ArduPilot/ardupilot #7707: Copter: move contents of commands_logic.cpp to mode_auto.cpp

rmackay9, 2018-02-13:
We've done a fairly major refactoring of the Copter code but there are still a few bits to be completed.

[commands_logic.cpp](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/commands_logic.cpp) has Copter::ModeAuto methods that should be moved to [mode_auto.cpp](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/mode_auto.cpp).

Sorry to be picky but when this is done, please ensure the order of the methods in mode_auto.cpp matches the order in the mode.h header file.

This should be a non-functional change I think.

bCamba, 2018-02-15:
This one seems simple. I am new to ArduPilot, so if it possible I want to take it.
@rmackay9 How do I test this after the code has been moved?

abhayjindal51, 2018-02-16:
@bCamba if you are not working on this issue, can I start working?

bCamba, 2018-02-17:
@abhayjindal51 I don't know what is ArduPilot on that, but I have already started working on this issue.

@rmackay9 I have some questions while doing this:

1 - You said to make sure the order of the methods of mode_auto.cpp match the mode.h header file. Is this the priority when moving the functions? Because I noticed there are some comment headers in commands_logic.cpp like "Verify Condition(May) Commands", "Do (Now) commands", etc...
And it seems as the order of the functions in mode.h sometimes break these headers apart. Let me illustrate this:

Excerpt from mode.h
```
    Location_Class terrain_adjusted_location(const AP_Mission::Mission_Command& cmd) const;

    void do_takeoff(const AP_Mission::Mission_Command& cmd);
    void do_nav_wp(const AP_Mission::Mission_Command& cmd);
    void do_land(const AP_Mission::Mission_Command& cmd);
    void do_loiter_unlimited(const AP_Mission::Mission_Command& cmd);
    void do_circle(const AP_Mission::Mission_Command& cmd);
    void do_loiter_time(const AP_Mission::Mission_Command& cmd);
    void do_spline_wp(const AP_Mission::Mission_Command& cmd);
#if NAV_GUIDED == ENABLED
    void do_nav_guided_enable(const AP_Mission::Mission_Command& cmd);
    void do_guided_limits(const AP_Mission::Mission_Command& cmd);
#endif
    void do_nav_delay(const AP_Mission::Mission_Command& cmd);
    void do_wait_delay(const AP_Mission::Mission_Command& cmd);
    void do_within_distance(const AP_Mission::Mission_Command& cmd);
    void do_yaw(const AP_Mission::Mission_Command& cmd);
    void do_change_speed(const AP_Mission::Mission_Command& cmd);
    void do_set_home(const AP_Mission::Mission_Command& cmd);
    void do_roi(const AP_Mission::Mission_Command& cmd);
    void do_mount_control(const AP_Mission::Mission_Command& cmd);
#if CAMERA == ENABLED
    void do_digicam_configure(const AP_Mission::Mission_Command& cmd);
    void do_digicam_control(const AP_Mission::Mission_Command& cmd);
#endif
#if PARACHUTE == ENABLED
    void do_parachute(const AP_Mission::Mission_Command& cmd);
#endif
#if GRIPPER_ENABLED == ENABLED
    void do_gripper(const AP_Mission::Mission_Command& cmd);
#endif
#if WINCH_ENABLED == ENABLED
    void do_winch(const AP_Mission::Mission_Command& cmd);
#endif
    void do_payload_place(const AP_Mission::Mission_Command& cmd);
    void do_RTL(void);
```

Functions under Nav(Must) comment header in command_auto.cpp
```
void Copter::ModeAuto::do_takeoff(const AP_Mission::Mission_Command& cmd)
void Copter::ModeAuto::do_nav_wp(const AP_Mission::Mission_Command& cmd)
Location_Class Copter::ModeAuto::terrain_adjusted_location(const AP_Mission::Mission_Command& cmd) const

void Copter::ModeAuto::do_land(const AP_Mission::Mission_Command& cmd)

void Copter::ModeAuto::do_payload_place(const AP_Mission::Mission_Command& cmd)

void Copter::ModeAuto::do_loiter_unlimited(const AP_Mission::Mission_Command& cmd)

void Copter::ModeAuto::do_circle(const AP_Mission::Mission_Command& cmd)

void Copter::ModeAuto::do_loiter_time(const AP_Mission::Mission_Command& cmd)

void Copter::ModeAuto::do_spline_wp(const AP_Mission::Mission_Command& cmd)

}

#if NAV_GUIDED == ENABLED
void Copter::ModeAuto::do_nav_guided_enable(const AP_Mission::Mission_Command& cmd)
#endif  // NAV_GUIDED

void Copter::ModeAuto::do_nav_delay(const AP_Mission::Mission_Command& cmd)

#if PARACHUTE == ENABLED
void Copter::ModeAuto::do_parachute(const AP_Mission::Mission_Command& cmd)
#endif

#if GRIPPER_ENABLED == ENABLED
void Copter::ModeAuto::do_gripper(const AP_Mission::Mission_Command& cmd)
#endif

#if NAV_GUIDED == ENABLED
void Copter::ModeAuto::do_guided_limits(const AP_Mission::Mission_Command& cmd)
#endif

#if WINCH_ENABLED == ENABLED
void Copter::ModeAuto::do_winch(const AP_Mission::Mission_Command& cmd)
#endif
```

It seems like this block from mode.h is the Nav(Must) block from the .cpp, but in a differente order. However, notice that in the .h file, functions start from the Nav(Must) block:
terrain_adjusted_location, do_takeoff, do_nav_wp, do_land, do_loiter_unlimited, do_circle, do_loiter_time, do_spline_wp, do_nav_guided_enable, do_guided_limits, do_nav_delay.

Then become exactly the Condition(May) block: do_wait_delay, do_within_distance, do_yaw

Then become **part** of the Do(Now) block: do_change_speed, do_set_home, do_roi, do_mount_control, do_digicam_configure, do_digicam_control

Then it goes back to the Nav(Must) block with do_parachute. This is why I got really confused. I am guessing either comment blocks or the order of mode.h files should be prioritized. Could you please elaborate on that?

2 - I noticed these functions are missing in both .cpp's (and are not defined in the .h).

```
spline_start(const Vector3f& destination, bool stopped_at_start, AC_WPNav::spline_segment_end_type seg_end_type, const Vector3f& next_spline_destination)
payload_place_run_release()
```

3 - There were some debugging flags in verify_payload_place(), shall I keep them?

4 - void nav_guided_run() had an if NAV_GUIDED statement in cpp but not in .h

bCamba, 2018-03-22:
Done and testes with SITL.
Pull request: #7991

peterbarker, 2019-01-14:
@bCamba Fixed this issue - thanks!

closed 2019-01-14 by peterbarker