ArduPilot/ardupilot #7710: Onboard OLED Display: enable for Plane

GrahamDyer, 2018-02-13:
#### Issue details
Enable the OLED onboard display in ArduPlane as described here: http://ardupilot.org/copter/docs/common-display-onboard.html for ArduCopter 

#### Version
All versions

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Fixed wing

#### Hardware type
AUAV-X1

#### Logs
N/A

magicrub, 2018-02-13:
Oh, this doesn't work for plane right now? I bought a few of those oleds but never got around to using them. I assumed they were working. Aaaand now I get to use them!

GrahamDyer, 2018-02-13:
Hi Tom, they work well in ArduCopter but the parameter NTF_DISPLAY_TYPE doesn't appear in Plane so I guess the code isn't there.

Naterater, 2018-02-14:
I thought precision hawk had the feature built in to their airframes. I never flew one though, so maybe I'm wrong, and maybe it was a special version.

magicrub, 2018-02-14:
@GrahamDyer There's nothign in the drivers that make copter special, it should work on all vehicles. I also see it in the auto-generated parameters list for plane [here](http://ardupilot.org/plane/docs/parameters.html#ntf-display-type-type-of-on-board-i2c-display) so the source says that parameter exists for plane so please double and triple check. All indicators say it should work equally on all vehicles.

GrahamDyer, 2018-02-14:
[image]
The NTF_DISPLAY_TYPE parameter isn't there. I'll load the latest version of AP and check again...

DavidIngraham, 2018-02-20:
Hmm, could this be specific to the AUV-X1? It works for me on the Cube.

GrahamDyer, 2018-02-20:
OK, I managed to get it working. Seems the version of AP that I was using didn't have it. Apologies. Thank you for looking at this.

closed 2018-02-20 by GrahamDyer