ArduPilot/ardupilot #7719: Sub: make RCMAP_FORWARD and _LATERAL to be sub only

magicrub, 2018-02-14:
Looks like param RCMAP_FORWARD and RCMAP_LATERAL are only used by Sub but they're being compiled in for all vehicles. It should probably only be compiled for Sub.

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RCMapper/AP_RCMapper.cpp#L48

Finders fee goes to [Nathan Eick](https://www.facebook.com/groups/ArduPilot.org/permalink/1719129351443584/). Nice one!

magicrub, 2018-02-14:
AP_GROUPINFO_FRAME is not being respected. All params that are flagged for specific build frames show up in all frame. Example, the BATT_WATT_MAX param is plane only, but shows up in copter and sub and rover wikis.

Naterater, 2018-02-15:
GND_SPEC_GRAV: Specific Gravity (For water depth measurement) is also in the Plane Wiki.

magicrub, 2018-02-16:
Turns out this is a global problem. Many params in all vehicles are being included that shouldn't be. There's no harm in it now, the params don't do anything, they just take up memory that they shouldn't and add confusion

Naterater, 2018-03-27:
Bump. It's still showing up in the wiki parameters lists too.

rmackay9, 2018-03-27:
This is a fairly easy thing to fix.  In the AP_RCMapper.cpp file linked above, we need to surround the Forward and Lateral parameters (including the descriptions above them) within a "#if APM_BUILD_TYPE(APM_BUILD_ArduSub)" and "#endif".  It's OK to leave the parameters defined within the AP_RCMapper.h header file.  So the parameter will actually still exist in memory but people won't see them in their parameter list.

jaxxzer, 2018-03-28:
> AP_GROUPINFO_FRAME is not being respected. All params that are flagged for specific build frames show up in all frame. Example, the BATT_WATT_MAX param is plane only, but shows up in copter and sub and rover wikis.

If this is the case, this is the real issue here. @rmackay9 This GROUPINFO_FRAME was designed to keep such #if #endif blocks to a minimum.

WickedShell, 2018-04-26:
Groupinfo frame is working correctly. The only issue of any note here is that the param doc generator doesn't handle it, that can be addressed, but the `#if APM_BUILD_TYPE` stuff should not be done, unless we decide we no longer want to use `AP_GROUPINFO_FRAME` or restrict the frame stuff to runtime changeable frame types.

jaxxzer, 2018-04-26:
Should this move to a new issue for the doc generator?

WickedShell, 2018-04-26:
I had the thought, but doc generator is part of this repo :) It is true that using `#if` would make a slightly smaller binary, but I thought the overall conclusion was to just us GROUPINFO_FRAME more. (If we want to limit the usage of GROUPINFO_FRAME to runtime selectable types that's a different discussion.

jaxxzer, 2018-04-26:
I've been discouraged from the use of the BUILD_TYPE, and if I remember right, the GROUPINFO_FRAME was introduced to address exactly the situation that would cover.

I did mean just a new issue in this repo with a new title and conversation.

PR #8737 (2018-07-17): AP_RCMapper: Mark sub specific parameters as sub only for docs
Intended to fix #7719

EDIT: Just wanted to note that in local testing this semed to cut the params from plane correctly, however I don't see any of the RCMAP params in sub.
closed 2018-07-17 by peterbarker