ArduPilot/ardupilot #7732: Arming

cody303, 2018-02-16:
The Ardupilot code reports the vehicle as armed when the safety button is not pushed, and the aircraft is not actually armed. This makes it so it is not possible to use the calibration functionality in QGCS if the safety button alone is used to arm. Would it be possible to have mavlink report the vehicle is unarmed until the safety button is pushed?

peterbarker, 2018-02-19:
Length discussion on gitter on this topic.

Resolution was basically that Don was going to work around the problem; have an `is_dangerous()` call which would return `is_armed() && safey_switch_off()`.

People are finding "armed but not dangerous" confusing.

olliw42, 2018-02-20:
I have read the gitter, and I totally agree with Don that with inventing a new word like dangerous you are going to much confuse the 99% of users he is mentioning, who would assume that armed means armed. This is just about wording, but often it's crucial to avoid confusion.
I think that instead of inventing a new word for the case (is_armed , saftey_switch_off) = (true , true) and to use armed for (true , false), one should wrap it around and invent a new word for the case (true , false) and use armed for (true , true). This aren't good ideas, but maybe something like "cold armed", "safe armed" etc.
I also would think that this should be reflected in the code, i.e. that is_dangerous() becomes is_armed(), and is_armed() becomes is_saftey_armed() (or whatever word you've chosen)
my 2 user cents

auturgy, 2018-02-20:
I agree with Olli and Don: armed should mean armed - ready to fly/drive in all respects. Clearly differentiating “soft arm” makes a lot of sense. 
Conversely, users shouldn’t be arming (hard or soft) until all setup is complete and they’re actually ready to fly, so this corner case could be resolved through better documentation and education.

peterbarker, 2018-02-20:
On Tue, 20 Feb 2018, olliw42 wrote:

> I have read the gitter, and I totally agree with Don that with inventing a new word like dangerous you are going to much confuse the 99% of users he is
> mentioning, who would assume that armed means armed. This is just about wording, but often it's crucial to avoid confusion.

I took notes on our discussion today in the dev call.  You should have 
received a copy of those by now.  Since I was involved in the discussion, 
the notes are probably of poor quality.  If you've got questions, 
please ask.

I think the most interesting thing to come out of it is that on some 
platforms the two "bits" (armed | safetyon) come from two different 
places.

One is definitely a main-MCU thing - "armed" is a high-level ArduPiloty 
thing.

safetyon is an IO thing; if the MCU dies the safety switch will still 
work.

To a certain extent that's becoming a bit of a curiosity....

olliw42, 2018-02-20:
thx for the clarification
I think I understand the different levels of origin, and that all four states are fully legitimate, and I didn't wanted to dispute that in any way. I myself don't see any problem with the current behavior. I was only trying to pledge to use a hopefully less confusing naming of the states. Using 'dangerous' for what should be the most common state is a bit confusing, IMHO. That's all I was trying to suggest. :)

Naterater, 2018-03-14:
This topic has been discussed on a few Dev Calls, and the resolution has been to stick to the current behavior and 4 legitimate states. Although using the word "dangerous" in the code may seem odd for a common state, end users should not see it. The solution provided via Gitter appeared to provide a solution.

closed 2018-03-14 by Naterater