ArduPilot/ardupilot #7737: Auto-Landing Throttle abort scaling is not correct when using reverse thrust

jmachuca77, 2018-02-18:
#### Issue details
When a plane is equipped with reverse thrust, the RC input for forward thrust is scaled differently than it normally is. That is 1500 is 0% throttle, 2000 is 100% throttle and 0 is -100% throttle. The landing abort code is reading the absolute value to determine wether it should abourt the landing so you end up aborting even if the throttle is lower than it should be.

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Planes

#### Hardware type
Pixhawk

#### Logs
Logs can be provided by Philip

Naterater, 2018-02-18:
@magicrub  is working on adding a reverse thrust PWM flag via another channel. It would be good to make sure the land abort code also works with it.

In this case, 1000 is 0% throttle, 2000 is 100% throttle and 2000 is -100% throttle (when another output channel is switched to PWM_MAX). https://github.com/magicrub/ardupilot/commit/c0586d1b0bd8f4bc6bad31cfd05e46d36cae3427#diff-3d04944173d87985779a1ba7b2f88b4bR123

jmachuca77, 2018-07-20:
@magicrub this is the issue that we found in Canberra with the X8

jmachuca77, 2019-01-09:
@magicrub would have to comment on this... I thought he already fixed this

IamPete1, 2021-09-12:
fixed by https://github.com/ArduPilot/ardupilot/pull/9743

closed 2021-09-12 by IamPete1