ArduPilot/ardupilot #7750: Plane: vertical displacement with QuadPlane

expntly, 2018-02-21:
#### Issue details
There doesn't seem to be a way to move the quadplane purely in the Z in AUTO.  All I'm looking for is a way to de-transition from fixed-wing to VTOL, have the vehicle descend **vertically** by N meters, loiter there for a while, climb back up **vertically**, and go back to fixed-wing.

I've tried a few different combinations, none of these seem to work:
* DO_VTOL_TRANSITION to VTOL and then navigate vertically with waypoints at the same lat,lon but different altitudes (opened another bug for an issue with this command)
* LOITER_TIME in Q_GUIDED_MODE=1: this effectively de-transitions the plane properly, but then WAYPOINTs that are close in lat,lon will get "skipped"
* LOITER_TO_ALT
* multiple WAYPOINTs at different altitudes;  also tried param1 and param2 to create a sphere, this only gets worse
* Q_ENABLE=2 (this seemed to work slightly better, but I couldn't get the vehicle to climb again much)
* CONTINUE_AND_CHANGE_ALT (apparently only makes sense for pure fixed wing)
* VTOL_LAND with an alt > 0 (bad idea, will disarm)

#### Version
3.8.2

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
SITL

#### Hardware type
SITL

expntly, 2018-03-17:
Anyone willing to look into this?  Does this even work with ArduCopter?  Thanks

lthall, 2018-03-18:
This is not a support forum.

closed 2018-03-18 by lthall