ArduPilot/ardupilot #7751: Using ArduPilot on ESP32 (with Xtensa LX6 Dual Core) based new designed Autopilot Hardware (Sigma2 Autopilot)

meminyanik, 2018-02-21:
#### Issue details
Our team has designed a new autopilot hardware based on ESP32 microcontroller (includes Xtensa LX6 Dual Core processor). Initial details of the hardware can be found at https://sidusinnovation.wixsite.com/sidusinnovation/sigmadual-autopilot. We started to write a basic autopilot software just for low level controls. But, we would appreciate if we can deploy ArduPilot software stack to this new designed autopilot.

#### Version
Final version.

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Quadcopter

#### Hardware type
We have designed a new autopilot hardware (Sigma2 Autopilot) based on ESP32 microcontroller. We would like to deploy ArduPilot software stack on this new hardware.

#### Logs
NA

auturgy, 2018-02-21:
Exciting!
I’d suggest that the ChibiOS port is where to start.
Jump on the ArduPilot ChibiOS gitter channel (I cross referenced this issue there). Be patient though, as there’s a bunch of stuff happening across different time zones.

auturgy, 2018-02-21:
As an additional comment: I see you are using the mpu6050. That imu is end of life and i2c only. I’d recommend considering a current generation (icm20789 or similar) imu, over spi.

meminyanik, 2018-02-21:
Thank you so much for your great comments. We are also developing the new version of the Sigma Autopilot and we will definitely consider mentioned IMUs on that board.
For deploying ArduPilot on ESP32, what about NuttX OS? ESP32 supports it. By using Arduino Core and NuttX together, can we use PX4 HAL for this modification?

auturgy, 2018-02-21:
ArduPilot is moving away from nuttx.
You could bring up your board on nuttx if you like, but I’d suggest ChibiOS is a better investment of your time.

Regards,

James

tridge, 2018-02-26:
as James says, we're moving away from NuttX on stm32 as we are getting much better performance on ChibiOS. I don't see a ChibiOS port to ESP32 yet though, so you'd either need to do a port or use NuttX or FreeRTOS. Any of those options would be quite a lot of work.
I'd suggest you talk to Giovanni about a port of ChibiOS to ESP32 as that would give you the best result.
You can hear more about the ChibiOS port of ArduPilot in my recent presentation here:
https://www.youtube.com/watch?v=y2KCB0a3xMg
(sorry, its a bit long)

meminyanik, 2018-02-26:
James and Andrew, thank you both for your great guidance. Then it will be a great chance for us to catch your technology by porting ChibiOS to our board. As you said, our initial plan was starting with FreeRTOS, but it would need lots of work. It will be awesome if we can port ChibiOS to ESP32 with the help of Giovanni, I will definitely contact him. Thanks a lot.

kostyamat, 2018-06-05:
@meminyanik 

To get a good start, you could port MultiWii firmware to your platform. It has not developed for a long time, but it will allow to connect more people who are interested in ESP32, but who continue to use the Arduino IDE, and do not want to move to complex console compilers. And to include more developers in the development, you could share the wiring diagram of the standard NodeMCU ESP32, and sensors such as GY-86, as is done on your PCB.

mkeyno, 2018-06-19:
@meminyanik  request @me-no-dev to help , he is porting FreeRTOS on ESP32 and he was one of the old fashion multiwii code developer

me-no-dev, 2018-06-19:
I'm here to help (and test) if you are interested ;)

davidbuzz, 2018-11-13:
personally, I'd love to see a chibios on esp32, and as a result ardupilot running on this platform, but the chibios port is beyond me.

meminyanik, 2018-11-13:
That would be great. I think, porting chibios to esp32 is the most challenging part of that process. I would be very happy to donate some of our flight control hardware if someone is interested. Our flight control sofware on https://github.com/DevSidus/Sidus_Github is working pretty stable. However, combining this powerful MCU and Ardupilot can provide perfect opportunities.

davidbuzz, 2018-11-13:
Giovanni has suggested he is open to the idea if a few of us can get together and work on it... see here... please add your name there if you are willing to help...
http://www.chibios.com/forum/viewtopic.php?f=3&t=3699

meminyanik, 2018-11-13:
Sounds great. I have replied. I would be very happy to help.

mkeyno, 2018-12-06:
@sh83 great job on ESP32 , have you design any hardware yet or test on any module?
I'm also working on customize  ESP32 HAL abstract with link to some 4G module but still working on hardware, also I believe FreeRTOS pretty mature and implemented yet for async core computation along with webserver services 
I hope you guys in this thread could have look   
https://github.com/mkeyno/4g_wifi_Ardupilot

sh83, 2018-12-07:
@mkeyno The first target is the simple esp32 dev module with external  10dof sensor connected.

sh83, 2019-02-17:
Current progress:
1. Ardupilot boot and work on esp32 devboard + gy-91 mpu9250/bmp280 board
2. Spi/wifi/work well
3. Mcu performance is ok to run 400hz loop. 800hz needs further optimization. @me-no-dev do you have working gprof support?

gadavidd, 2020-05-18:
Can you conected esp32 Lora with pixhawk like a Telemetry?

davidbuzz, 2020-05-19:
> Can you conected esp32 Lora with pixhawk like a Telemetry?

Im sorry, but i don't understand your question....  Lora is a Radio technology, and has nothing to do with esp32.    i don't think this issue is the right place for your question.

kd0aij, 2020-09-02:
Interesting: https://www.hackster.io/rayburne/lora-and-the-esp32-6ce9ba

love-bytes, 2020-11-07:
@sh83 
> 
> 
> Current progress:
> 
>     1. Ardupilot boot and work on esp32 devboard + gy-91 mpu9250/bmp280 board
> 
>     2. Spi/wifi/uart work well
> 
>     3. Mcu performance is ok to run 400hz loop. 800hz needs further optimization. @me-no-dev do you have working gprof support?
Hey, this is some impressive development. What is your current process?

ksj20, 2021-07-14:
Any updates on this? Where can I find the details for this project?

auturgy, 2021-07-14:
There’s an esp32 dev channel in discord: https://ardupilot.org/discord

davidbuzz, 2021-07-14:
https://github.com/davidbuzz/ardupilot/blob/esp32_master_new_flat/libraries/AP_HAL_ESP32/README.md

. .both myself and @sh83 have branches that work with gy-91 imu, and generally they are OK for developers to hack on. We've flown stuff with it. No one has made any production runs of Pcbs etc, just development stuff... but its at the point where someone totally could.

ksj20, 2021-07-14:
Thanks for the heads up, appreciate the help

davidbuzz, 2023-12-19:
ESP32 is experiemental and has known limitations and is not suitable for someone who has no experience on linux, it needs a custom board configuration for every diy board that's made, and also needs to be custom-compiled for these, so there's "no point" releasing binaries, as there's no 100% "standard" configuration.  The README here: https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_HAL_ESP32 documents the build process, and *one* common pinout for people comfortable with linux, so they can build their own, but its not in any way100% ready.   Either the README  and the 'esp32' discord channel together is is enough information for you to to have-a-go with the wiring/configure/build/flash process, or its not.

davidbuzz, 2026-06-15:
that configuration is *not* in alighment with the original 'buzz' pinout, so it should have its own board name, id recomment esp32_davut or your preferred firstname, "buzz" happens to be *my* preferred name. :-)