ArduPilot/ardupilot #7753: AP_UAVCAN: add esc.Status and store data in dataflash

olliw42, 2018-02-21:
in the link below are for download new versions of AP_UAVCAN.h and AP_UAVCAN.cpp, derived from today's master, with these additions:

* support for digesting the esc.Status UAVCAN data type
* storing of the received motor Status data in the respective DataFlash type LOG_ESCx_MSG 

link: www.olliw.eu/drop/ap/AP_UAVCAN-wescstatus.zip
please search for //OW to easily find the additions

I do have this pieces of code in use since 2 weeks, and did multiple test flights since then (https://www.rcgroups.com/forums/showpost.php?p=39138712&postcount=501).

In some future one could of course make better use of the esc.Status message, but this approach at least allows one to analyze the motor telemetry after flight already now, by using e.g. MissionPlanner, which IMHO is useful. 

have fun, Olli

EShamaev, 2018-02-22:
Can you make a PR with this changes?

auturgy, 2018-02-22:
I've removed some of the unneeded code and comments, then pushed it to a branch here: https://github.com/ArduPilot/ardupilot/tree/OlliW_UAVCAN_ESC_FB so it can at least be looked at.  If I'm really lucky, Olli will play with it in his browser, and hopefully make the PR :)

olliw42, 2018-02-22:
many thx, sir!

this line https://github.com/ArduPilot/ardupilot/blob/OlliW_UAVCAN_ESC_FB/libraries/AP_UAVCAN/AP_UAVCAN.cpp#L380 would have to go, I think

as regards the removed comments: 
They kind of "marked" code pieces which at a later stage would be required, but since they are not currently used they are not really testable. So, it could be useful to "have them at hand" to lessen the effort then they are needed,  but I wouldn't promote it to "active" code. I have no idea what your procedures for such situations are. I tried to "resolve" it with the comments. 
(I didn't left them just to make life harder ;))

I don't understand what a PR is at this point, I mean it's in the repo and just needs merge (in case it would be found acceptable of course)

I'll try to compile this here.

olliw42, 2018-02-22:
results of bench tests, with the line removed: looks all good
in more detail:
I compiled ArduCopter for v4, no issues
flash a "naked" pixracer (no gps, no ext mag etc), with 4 uavcan esc nodes connected to the can bus: no issues with the esc status 
here the .bin: www.olliw.eu/drop/ap/49-01.01.2000-01-05-20-uavcanescstatusbranchtest.bin
you'll find the ESCx fields

auturgy, 2018-02-26:
Thanks Olli.
I removed that line.
I’m not trying to be rude about PR’s and such: I put your code in a branch so you could make the pull request from the browser, and hence be linked to your work in the git history - my phrasing was perhaps too flippant.
You are doing good things, and it would be nice to see that credited.
I can make the PR if you’d prefer.

olliw42, 2018-02-26:
thx
as regards PRs, it's not really a question of preference since I'm still miles away of doing such things, so someone else doing it would momentarily be the only way
as said, it's all your choice, I just suggest, and you're free to throw it away or use it, or whatever else you want to do with it :)

olliw42, 2018-02-26:
ok, looks good
compiled the branch for v4, no issues
loaded onto pixracer with rudimentary setup, with 4 uavcan ESCs, disarmed and let the motors spin by moving the sticks, and data are in the log's ESCx fields
here's the bin: www.olliw.eu/drop/ap/56-01.01.2000-01-05-26-uavcanstatusbranchtest-no2.bin

Naterater, 2019-03-03:
What's holding this up? I can test now too if you build me a quick plane cube black binary.

Naterater, 2019-09-29:
Any chance this could happen? More people are trying to take advantage of the data that is currently being thrown away. https://forum.zubax.com/t/kotleta-20-can-bus-telemetry-data-for-rpm-in-log/1433/2

auturgy, 2019-09-29:
Not far away 

Regards,

James

tridge, 2020-01-04:
fixed in #13164

closed 2020-01-04 by tridge