ArduPilot/ardupilot #7756: Lightware Serial driver does not handle bad/out of range or slow readings

RickReeser, 2018-02-21:
I fly a lightly modified Copter 3.4.x, but the driver is the same in Master. Applies to all platforms.

I have encountered two problems with the Lightware serial driver. First, it does not handle invalid readings from the sensor properly, and second, it has no provisions to handle readings from the sensor that are slower than the loop rate that calls get_reading.

## Bad readings handling
The Lightware SF/20 sends a value of -1 when it is out of range or fails to get a good reading. The serial driver's ascii-to-float function does not look for negative signs when it interprets the serial data from the sensor. This means it reads -1 as 1, which leads to two problems:

1. Because the measurement used by Ardupilot is an average of multiple readings, the average will be incorrect if it included any bad readings. Instead, the bad readings should not be included in the average.
2. Because 1 is a valid reading (as long as it is more than RNGFND_MIN_CM), if the rangefinder is out of range, it will produce a constant value of 1, causing the "rangefinder ok" flag to be true with a reading of 1 meter while the aircraft is in the air!

The current code for get_reading() [is this](https://github.com/ArduPilot/ardupilot/blob/ab413b46058ccb83527d36414a62e1e734028186/libraries/AP_RangeFinder/AP_RangeFinder_LightWareSerial.cpp#L56-L67):

    float sum = 0;
    uint16_t count = 0;
    int16_t nbytes = uart->available();
    while (nbytes-- > 0) {
        char c = uart->read();
        if (c == '\r') {
            linebuf[linebuf_len] = 0;
            sum += (float)atof(linebuf);
            count++;
            linebuf_len = 0;
        } else if (isdigit(c) || c == '.') {
            linebuf[linebuf_len++] = c;

I tried the following, which only includes the reading into the average if the number from atof() is positive:

    float sum = 0;
    uint16_t count = 0;
    int16_t nbytes = uart->available();
    while (nbytes-- > 0) {
        char c = uart->read();
        if (c == '\r') {
            linebuf[linebuf_len] = 0;
            float tmpnum = (float)atof(linebuf);
            if (tmpnum > 0) {
                // only average if reading is > 0, otherwise ignore
                sum += (float)atof(linebuf);
                count++;
            }
            linebuf_len = 0;
        } else if (isdigit(c) || c == '.' || c == '-') {
            linebuf[linebuf_len++] = c;

I have been flying with this modification for months now, and I haven't encountered any further issues with my SF/20. It's a pretty naive modification, though, so maybe there are corner cases that I didn't think of.

## Slow readings handling

I actually encountered this issue when using the LW serial driver for a non-LW device which I had set to a slower publish rate. I think it is still relevant to the LW driver because all LW devices have a user-configurable update rate, so I think it will still be possible for users to encounter this problem with some configurations.

Currently, if the sensor does not produce any readings in time to be included in the last get_reading() call, `sum` will be 0, resulting in a reading of 0. A sensor that has a slower update rate than the loop rate will produce spiky readings that bang between 0 and the real reading.

I'm not sure if this case should be handled by the driver, but it may be appropriate to use the last valid value for a limited time if `count` is 0. Or perhaps there is a way to change how often get_reading() is called so that it can match the sensor's rate.

One potential problem here is with the above change to handing bad readings, it will be possible for `count` and `sum` to be 0 when the lidar gets a few bad readings in a row, since the -1's would be thrown out. That shouldn't be a problem (a reading of 0 is an okay way to record a bad reading), but we may need to differentiate the cases of bad readings vs. no readings if we want to handle the no readings case.

rmackay9, 2019-08-03:
I think both these issues are resolved now.  The slow reading handling should be fixed because we use a 200ms timeout to decide whether to set the status to no-data or not.  I have not specifically tested this though so please tell me if I'm wrong and we can try and address it.

closed 2019-08-03 by rmackay9

wynn4, 2021-09-07:
I believe there is still a bug with the LightWare Serial driver when it comes to handling out-of-range/lost signal outputs from the lidar--specifically when using the SF11.

The SF11 by default reports a value of 130.0 when the lidar signal is lost, and the current driver treats this as a valid measurement since it only filters out measurements that are less than zero. Since the driver averages "valid" results, you can have a situation where:
Rreading_1 = 130 (signal lost, not a good reading)
Reading_2 = 10 (good reading)
Reading_returned = 70 (bad reading skewed by the 130 value)
In the above if the user is using the SF11 / C and the RNGFND_MAX_CM is set to 12000 as the docs suggest, then this will be interpreted as a good reading.

At a minimum, I think that the value 130 needs to be flagged as an invalid result. 130 meters is longer than the range of almost all of LightWare's products ranges so I don't think this extra check would affect other devices that use this same driver.

rmackay9, 2021-09-07:
@wynn4,

Thanks for this.  I don't think we knew that the lidar output 130m when they lose their signal.  I think we thought they always output a negative value.

.. but here's the line backing up what you say in [the SF11 datasheet](https://www.documents.lightware.co.za/SF11%20-%20Laser%20Altimeter%20Manual%20-%20Rev%2010.pdf):

[image]

Unfortunately this means we need to go through all the lightware datasheets for other products that use the same serial protocol to see if they are consistent (I'll bet they're not!).

rmackay9, 2021-09-07:
I've created a new issue here https://github.com/ArduPilot/ardupilot/issues/18602