ArduPilot/ardupilot #7772: RTCM inject from antenna tracker

suraqis, 2018-02-23:
#### Issue details
Now it seems that no way to inject RTCM messages from GPS RTK enabled base module connected to antenna tracker. It would be handy to receive correction within MAVlink from antenna tracker.

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[ X ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

geofrancis, 2018-05-20:
this would be great, mission planner crashing is a PITA.

proficnc, 2018-05-20:
This would be awesome