ArduPilot/ardupilot #7773: Tradheli Problem after upgrade from copter 3.5.4 to 3.5.5

Spirosta, 2018-02-23:
#### Issue details
I was running tradheli 3.5.4 with the throttle set as setpoint mode and setpoint to 1425.

H_RSC_MODE | 2 |   | 1:Ch8 Input  2:SetPoint  3:Throttle Curve | Determines the method of rotor speed control
H_RSC_SETPOINT | 1425 | PWM | 0 1000 | PWM in microseconds passed to the external motor governor when external governor is enabled

At ch8out at version 3.5.4 it gives out the right setpoint of 1425.
But at version 3.5.5 is gives out whatever is set on SERVO8_MAX ignoring the setpoint of 1425

SERVO8_FUNCTION is checked that is 31

Why is this happening? is a bug of  V3.5.5?
As first time posting a HUGE THANK YOU to all developers for their time and effort to keep the project alive!

#### Version
APMcopter V3.5.5 mission planner 1.3.52.10

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
traditional heli

#### Hardware type
pixhawk 2 the cube from proficnc

#### Logs
_Please provide a link to any relevant logs that show the issue_

Spirosta, 2018-03-02:
Does anyone other confirm this bug?

auturgy, 2018-03-02:
Relevant: https://discuss.ardupilot.org/t/problems-with-throttle-channel-8-on-pixhawk/25967/18

ChristopherOlson, 2018-03-28:
This was determined by parameter analysis to be a problem in the Trad Heli setup page in Mission Planner causing an essentially broken setup, and not a problem in the code for Copter. I am going to close this issue here as nothing can be done in the Copter code for this problem. We will attempt to identify all the problems with the MP heli setup page and work with Micheal on either fixing it in MP, or removing that setup page and using only the Full Parameter List for heli setup.

closed 2018-03-28 by ChristopherOlson