ArduPilot/ardupilot #7782: Minor problems in building Ardupilot for target f4light_AirbotV2

anbello, 2018-02-24:
#### Issue details
If I build for Revolution with:
$ cd ArduCopter
$ make f4light
is all OK

if I try to build for AirbotV2 with:
$ make f4light BOARD=f4light_AirbotV2
I have some errors

with the following action errors disappear and building is OK:

[starting from ArduCopter dir]
$ cd ../libraries/AP_HAL_F4Light/hardware/osd/osd_core
$ ln -sf ../../../support/minimosd-extra/MinimOsd_Extra/Params.h Params.h
$ cd ../../../../../ArduCopter/
[a wrong symlink]

then edit the file: …/libraries/AP_HAL_F4Light/hardware/osd/osd_core/GCS_MAVLink.h
and change BetterStream in AP_HAL::BetterStream on lines 23, 26

then edit the file: …/libraries//AP_HAL_F4Light/hardware/osd/osd_core/OSD_Max7456.h
and change BetterStream in AP_HAL::BetterStream on line 88

#### Version
commit 9cffa13

#### Platform
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type

#### Hardware type
Omnibus F4 V2 Pro (AirbotV2)

#### Logs

night-ghost, 2018-02-26:
One souldn't to change anything in symlinked files in osd_core folder because they belongs to OSD submodule.

anbello, 2018-02-26:
OK I understand but there are difference between `Params.h`:
`/mnt/disk_d/src/quad-copter/minimosd-extra/MinimOsd_Extra/Params.h`
and all the other files in the dir `libraries/AP_HAL_F4Light/hardware/osd/osd_core`, for example `Panels.h`:
`../../../support/minimosd-extra/MinimOsd_Extra/Panels.h
`

PR #7792 (2018-02-26): AP_HAL_F4light: fixes OSD compilation issue
fixes #7782
closed 2018-02-26 by tridge