ArduPilot/ardupilot #7786: Plane: automatic landing with side wind for VTOL Tail Sitters

robustini, 2018-02-25:
#### Issue details
It is known that the "Tail Sitters VTOL" suffer from wind direction during automatic landing.
In the "non-vectored" versions the pitch control in Copter mode is quite precarious without the motors tilt system, while the roll using the revs variation of the motors is quite reactive and allows excellent corrections of the attitude.
What I propose is therefore the possibility of making automatic landings by rotating the Tail Sitters before the automatic descent so that it has the side wind.
It is well known that Ardupilot recognizes the wind direction after a few turns in plane so I don't think it's a big deal to implement this feature, and I think it is very important for a perfect ground touch.

#### Version
Plane V3.8.5 Beta 1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Tail Sistters VTOL

palm369, 2018-02-25:
Absolutly. And also need some logic which instantly kills throttle if plane falls over while landing.

IamPete1, 2022-03-06:
Closed by https://github.com/ArduPilot/ardupilot/pull/19859

closed 2022-03-06 by IamPete1