ArduPilot/ardupilot #7787: RTL_ALT 325m limit ??

RQRobert, 2018-02-25:
#### Issue details
I have found the RTL_ALT limit to 32767 (So it becomes 327 metre) .  It cant exceed this limit. 
I want this to be 500metre in my Pan.
Kindly extend the limits of this Parameter so i can do my Mission.

#### Version
_What version was the issue encountered with_
AC3.5.2

#### Platform
[  ] All
[  **] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
N/A

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
N/A

#### Logs
_Please provide a link to any relevant logs that show the issue_
N/A

rmackay9, 2018-02-28:
Yes, it appears we store the parameter as an AP_Int16 (search Parameters.h for rtl_altitude).  To fix this we need to replace this parameter with a new parameter which is of type AP_Int32 or AP_Float (probably float is better).  I'll provide some details on how this can be done:

- remove the rtl_altitude parameter definition from Parameters.h and Parameters.cpp
- rename the k_param_rtl_altitude enum to k_param_rtl_altitude_old in Parameters.h
- add rtl_altitude parameter definition to ParametersG2 (in Parameters.h and Parameter.cpp)
- all though the code, replace rtl_altitude with g2.rtl_altitude
- add to the conversion_table array in Parameters.cpp so that the old parameter value is copied to the new parameter.
- test the parameter conversion works in SITL by loading an old version of the firmware (like Copter-3.5.5), change the old RTL_ALT parameter, then load the new firmware and ensure the new value apppears in the new RTL_ALT parameter.  Note that this conversion only happens once so you might need to wipe the eeprom in order to perform the test more than once.
- test in SITL That the copter returns at the new RTL_ALT value

RQRobert, 2018-03-06:
Thankyou very much for the guidance @rmackay9 .

unfortunately, I am not a developer .
Kindly add this change in the next release .

rmackay9, 2018-03-06:
@RQRobert,

Recently I've been adding more detailed advice to nearly all the issues I raise as a way to help other developers who might want to pick up the issue.  I can't immediately make any promises as to when it will be resolved.  To add some detail on how things work, there are a few ways in which issues can be resolved:

- critical safety issues are often resolved by the maintainer (i.e. me, tridge, etc)
- a volunteer developer is interested/motivated to fix the problem and resolves it
- a company wants it fixed and either assigns one of their developers or [hires a contractor](http://ardupilot.org/ardupilot/docs/common-commercial-support.html) to resolve it

PR #18668 (2021-09-15): Copter: convert RTL_ALT to int32
Closes #7787

This allows a user to set a RTL_ALT higher than the previous max. of 326.67m which could be limiting depending upon terrain. 

Tested that it retains the user param value from master to this PR.
Tested that if the param is changed after the PR the param value is kept.
Tested in SITL that a RTL_Alt of  85600 cm is achieved before heading home.

Note I had to also change the RTL_CONE to 10 to get it to go above 32767 since my WPs were so close. 

Picture Below shows climb at current location to RTL_ALT, moves to RTL location at RTL_ALT, and then des [... cut]
closed 2021-09-15 by rmackay9