ArduPilot/ardupilot #7796: Rover: path between waypoints should use previous waypoint as origin

rmackay9, 2018-02-27:
In Rover-3.2.1 the path that the vehicle follows between waypoints has an "origin" which is the vehicle's position at the moment it entered the waypoint-radius around the previous waypoint.  This should be changed so the origin is the previous waypoint.

[image]

khancyr, 2018-02-27:
yep the origin for the nav_controller path calculation should be the real wp1 and not the point we consider as "we pass on the waypoint"
A nice addition could be to have an accuracy estimation so the rover could dynamically adjust the wp_radius and try to go as close as possible to the real waypoint coordinate without overshooting much

rmackay9, 2018-03-14:
This is fixed in master and will go out in the next release.

closed 2018-03-14 by rmackay9