ArduPilot/ardupilot #7824: Altitude can't be held when rangefinder enabled but not connected

Gangti, 2018-03-02:
#### Issue details
Hi
Recently I enabled rangefinder through parameters setting, and the copter works fine when range finder connected. 
But, if I disconnect the range finder, the copter's altitude couldn't be held in ALT_HOLD mode.
The proper feature is that barometer(for example) should be used when range finder enabled but not connected.

I reviewed the following code, and found that it fall into the logic "before takeoff we assume on-ground range value if there is no data" for takeOffDected is always false(note:I didn't use optFlow).

`libraries\AP_NavEKF2\AP_NavEKF2_Measurements.cpp
void NavEKF2_core::readRangeFinder(void)
......
// find the median value if we have three fresh samples
if (sampleFresh[sensorIndex][0] && sampleFresh[sensorIndex][1] && sampleFresh[sensorIndex][2]) {
......
} else if (!takeOffDetected && ((imuSampleTime_ms - rngValidMeaTime_ms) > 200)) {
                // before takeoff we assume on-ground range value if there is no data
                rangeDataNew.time_ms = imuSampleTime_ms;
                rangeDataNew.rng = rngOnGnd;
                rangeDataNew.time_ms = imuSampleTime_ms;

                // don't allow time to go backwards
                if (imuSampleTime_ms > rangeDataNew.time_ms) {
                    rangeDataNew.time_ms = imuSampleTime_ms;
                }

                // write data to buffer with time stamp to be fused when the fusion time horizon catches up with it
                storedRange.push(rangeDataNew);

                // indicate we have updated the measurement
                rngValidMeaTime_ms = imuSampleTime_ms;
}

`

#### Version
Latest master version.

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Copter/Hexa

#### Hardware type
Pixhawk2

#### Logs
Nothing

peterbarker, 2021-02-03:
This was fixed with https://github.com/ArduPilot/ardupilot/pull/14375 (and was backported to Copter 4.0)

@amilcarlucas for visibility

closed 2021-02-03 by rmackay9

amilcarlucas, 2021-02-03:
I think only the "flyaway fix" part of that PR got backported. The rest was deamed not stable enough for backporting