ArduPilot/ardupilot #7834: copter will not climb above certain altitude (~30m)

maciek01, 2018-03-04:
#### Issue details

copter will not climb above certain altitude (~30 m - cannot tell precisely - see log) in loiter and poshold mode. This is my first time testing with 3.5.5. My geofence is disabled.

unrelated issue:
i also observed a drift despite of servos all being at 1500, third attempt at RTL caused a flyaway, first and second attempt was fine.

#### Version

3.5.5

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type

HEX 680mm

#### Hardware type

pixhawk 2.1

#### Logs

https://drive.google.com/open?id=1VxF3pFrolqr2_Hm77EyynOckogxXSJU6

rmackay9, 2018-03-08:
Thanks for the report although this is not really the place for support.  Normally we ask that support requests be put in the forums.  Sorry to be strict but if we're not then the developer to-do list just turns into another support forum.  Getting answers to support questions can be difficult but remember that here and in the forums, this is all unpaid volunteer support so you can't set expectations too high really.

With that aside, I"m pretty sure this is all related to the optical flow sensor and accompanying lidar.  It's not letting the vehicle climb about 32m because it's using the optical flow for position control but it can only do that if the vehicle stays within the range of the lidar.  I suspect if you turn off the optical flow and only use GPS it will be able to climb higher.

It should be possible to use both optical flow and GPS and climb above the optical-flow/range-finders range at which point it should only use the GPS but I haven't personally tested this much.

Of course you've done the optical flow calibration mentioned on the wiki?

As a side note, the changes between Copter-3.5.4 and Copter-3.5.5 are quite small and not in this area of the code so I suspect Copter-3.5.4 would have behaved the same.  You could try moving back to 3.5.4 by using MP's Install Firmware screen's "Pick previous firmware" link (it goes back one version to 3.5.4) but I don't think it will help.

The log has a lot of EKF complaints.  Some as it switches between the first and second compass, and it's also complaining about the optical flow a bit I think.

Maybe you can try the above and the next step is for us to talk to Paul Riseborough to see if he can give more info using the logs.

maciek01, 2018-03-08:
Thanks so much. what is the URL for the support forum if you dont mind. I'm happy to move it there
Yes i did all the FLOW calibration as prescribed. Alt climb used to work fine with just LIDAR on one of the prior versions of the software. This is the first time trying to climb higher using FLOW and also the first time when positioning and RTL dont work as well.
Previous 5 flights on prior firmware and the same hardware  configuration (lidar+flow) were just perfect in all aspects (except i never tested alt >30m). thansk again.
Minor note auto-tune yielded "floaty" yaw control but its a different issue.

magicrub, 2018-03-08:
>what is the URL for the support forum

discuss.ardupilot.org

maciek01, 2018-03-13:
moved it here:

https://discuss.ardupilot.org/t/copter-will-not-climb-above-certain-altitude-30m/26834

closing as advised

closed 2018-03-13 by maciek01