ArduPilot/ardupilot #7838: CAMERA_FEEDBACK_FLAGS is unusable

WickedShell, 2018-03-05:
#### Issue details
CAMERA_FEEDBACK_FLAGS is meant to provide a bitmask of options for CAMERA_FEEDBACK, but it is not defined as a bitmask, rather it is defined as a strictly increasing enum. The current form is extremely limiting and burns the 8 bits of space on duplicate fields. I propose the following new structure (outlined in bit positions):
0. Video (if not set then it is a still photo)
1. Bad Exposure
2. Closed Loop (if unset then assume open loop camera)

It's worth noting that I think both the bit 0 and bit 1 values could be removed entirely for now, but since someone at least wanted them before I was trying to preserve them.

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

amilcarlucas, 2018-03-06:
Please remember to use the `display=bitmask` attribute.

WickedShell, 2018-03-06:
Conclusion from dev call was to rearrange as needed, and drop bad exposure.

amilcarlucas, 2018-04-07:
@WickedShell do you have a PR ? Should I do one ?