ArduPilot/ardupilot #7839: Copter/Sub: log message number should be an enum

rmackay9, 2018-03-05:
In Copter and Sub we define the dataflash log message ids using "#define" but we should instead use an enum like we do in Plane.  The reason to do this is that it makes maintenance of the list easier (it is less likely someone will pick a number that has already been used) and the specific number used for a log message is not important because at the top of each dataflash log file we tell the client (i.e. the ground station graphing tool) the id, name (i.e. "ATT"), fields, units, multiplier and format so there shouldn't be any hardcoding in the ground stations based on the specific id used in the vehicle code.

here's how it's done in [copter's defines.h](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/defines.h#L313).
here's how it is done better in [plane's defines.h](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/defines.h#L132).

EbinPhilip, 2018-03-11:
In defines.h in ardusub, LOG_PROXIMITY_MSG is being assigned value 36
https://github.com/ArduPilot/ardupilot/blob/1dd14a8aedf8d295ce04af857b8b8688080a90f2/ArduSub/defines.h#L126
But it was declared initially in LogStructure.h in Dataflash library with a different value as part of enum LogMessages
https://github.com/ArduPilot/ardupilot/blob/1dd14a8aedf8d295ce04af857b8b8688080a90f2/libraries/DataFlash/LogStructure.h#L1541
Now when I change the message ids in defines.h to an enum, i get an error, since LOG_PROXIMITY_MSG is being redeclared.
Since you mentioned that the no need not be specific, shall i remove the declaration in defines.h ?

rmackay9, 2018-03-12:
@EbinPhilip, I wasn't aware of the duplicate definition, I think you can just remove the one that isn't being used.  The bug probably happened when we moved the proximity message from the Copter code (where it was used originally) into the shared logging code.  Thanks!

rmackay9, 2018-03-14:
Fixed by @ EbinPhilip, thanks!

closed 2018-03-14 by rmackay9