ArduPilot/ardupilot #7848: Trigger camera by Mavlink

Vavooon, 2018-03-05:
#### Issue details
Currently camera can be only triggered using servo or relay. We're working on software which allows to control cameras using Mavlink and this feature stops us from using ArduPilot

#### Version
_Latest_

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_All_

#### Hardware type
_All_

amilcarlucas, 2018-04-07:
then you simply connect the tx and gnd pin from your telemetry modem to the rx and gnd bins of both pixhawk and the camera. This way both pixhawk and the camera will get the telemetry stream from the ground station, and you can control the camera.

You will however, not be able to get the camera replies sent to the gcs

Vavooon, 2018-04-07:
Thank you, however it wouldn't allow AP to make photos while flying a mission.
Meanwhile I've implemented the feature and can share a patch but since I have no experience with ArduPilot code base I'm not sure it is implemented in a good way. Couldn't you check it? I can ether post my changes here or create a PR.

WickedShell, 2018-08-03:
@Vavooon You have to configure a picture taking method, but we do emit a `COMMAND_LONG` with the command of `MAV_CMD_DO_DIGICAM_CONTROL` with param5 set. This is a MAVLink camera trigger. Is there a specific issue with this? (I realize the feature isn't advertised, but it is present)

Vavooon, 2018-08-08:
@WickedShell Thank you, probably I was waiting for MAV_CMD_IMAGE_START_CAPTURE cmd.
Will check if it works on my configuration soon.

WickedShell, 2018-08-08:
@Vavooon Yeah we don't currently support any form of `MAV_CMD_IMAGE_START_CAPTURE`.

jmachuca77, 2018-08-16:
I can cofirm that we currently emit the digicam_control message, we use it for triggering our smart camera since a few years ago. So closing this issue.

closed 2018-08-16 by jmachuca77