ArduPilot/ardupilot #7852: Expanding tailsitter support to 4 motors

rflowers, 2018-03-06:
#### Issue details
Expanding tailsitter support from dual motor support, to four motors supporting differential thrust in pitch and yaw.

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Tailsitter unique build (already manufactured)

#### Hardware type
Beaglebone Blue (primary) / Pixhawk (backup)

As far as I can see Arduplane currently only supports dual motor tailsitter. I seek to expand this to four motors in an X formation. I am starting the testing with flying as a quadcopter and doubling up the dual motor tailsitter code to get differential yaw. Ultimately I seek to expand the code to cover both pitch and yaw differential thrust.

Apart from AP_MotorsTailsitter, does anyone know any other critical pieces of code needed to support this? Any ideas/suggestions how best to implement this?

many thanks

samuelctabor, 2018-03-08:
Ideally would be really nice to separate the tailsitter element from the motor configuration. Then any of the Arducopter motors layouts could be used as a tailsitter. Perhaps AP_MotorsTailsitter could be renamed to AP_MotorsBicopter, and a parameter Q_TAILSITTER added to trigger the AHRS view rotation and the transition behaviour?

fordyi, 2018-04-02:
Also interested in this, using an x formation tailsitter. is there any code that you guys have written to test or any psudo-tutorials available for modifying the arducode?

magicrub, 2018-04-02:
+1 I'm interested in this too

kd0aij, 2018-10-26:
@fordyi Sorry for the very late reply...
This commit https://github.com/kd0aij/ardupilot/commit/8d3f221b4149d6573f60cecd3f433d6f272e9ecf adds differential thrust for pitch in a quad motor tailsitter. Note that the motor configuration is "+" though, not "X".  The + configuration has potential structural and cruise efficiency advantages over the X configuration. It would also be possible to support a trimotor configuration with this approach, to further reduce weight and complexity.

mojtabahdpr, 2019-01-06:
+1 I need confog for multirotor tailsitter

yuhahaha, 2019-01-08:
+1 Interested in using 4 motors for tailsitter

kd0aij, 2020-06-19:
"copter tailsitters" are now supported: see https://ardupilot.org/plane/docs/guide-tailsitter.html?highlight=copter%20tailsitter#tailsitter-configuration

closed 2020-06-19 by kd0aij