ArduPilot/ardupilot #7853: gps fuse error

liang-tang, 2018-03-06:
#### Issue details
[image]

[image]

#### Version
3.8.4

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ * ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
quadplane

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

liang-tang, 2018-03-06:
@priseborough

priseborough, 2018-03-06:
Please provide a link to your flight log.

liang-tang, 2018-03-07:
here is the link
https://drive.google.com/file/d/1ur8YvJdgiL2j6a1g4urNe1h7oqq2HYJi/view?usp=sharing

liang-tang, 2018-03-07:
another log，same issue，both of them， at the end of log 
[log2 link](https://drive.google.com/open?id=10FSemdYmC3wyMsGeEGu_FkgQ7lLpfGpA)

liang-tang, 2018-03-07:
[image]

closed 2018-06-22 by liang-tang