ArduPilot/ardupilot #7856: ArduSub : conflit between code and comment

khancyr, 2018-03-06:
#### Issue details
At https://github.com/ArduPilot/ardupilot/blob/master/ArduSub/defines.h#L242 , the defined value and comments doesn't match. on Copter, the timeout is on 5s. What should be the correct value for Sub ?

magicrub, 2018-03-06:
I think it's safe to assume the comment is always less accurate than the code. There are exceptions of course, so we'll want to run it by @rmackay9 

As far as Sub goes, I have no idea what data links you have and how tolerant you are of allowing loss of link before you get worried. I would guess it would be a longer number than copter but that's your decision.

jaxxzer, 2018-03-06:
The solution for sitl/speedup is to disable gcs failsafe, or increase gcs heartbeat frequency.

commit c1c471db (2018-03-06): Sub: Fix comment

Fix #7856
closed 2018-03-06 by jaxxzer