ArduPilot/ardupilot #7859: External compass orientation separation from AHRS orientation

Naterater, 2018-03-06:
#### Issue details
Right now, COMPASS_ORIENT is combined with AHRS orientations (According to the complete parameter list wiki). I think it should be separated for EXTERNAL compasses and reference the frame rather than the flight controller.

For internal compasses, the compass orientation parameters could be ignored and reference the AHRS_ORIENTATION so users can't mess that up.

In addition, the reverse transformations from the flight controller could get complicated if the AHRS orientations are expanded to include custom orientations as was proposed in the dev call on 3/5/2018.

#### Version
All

#### Platform
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

Naterater, 2018-03-06:
It may just be a documentation issue. Either way, it should be clarified.

[image]

rmackay9, 2018-06-30:
Yes, I'm 99% sure that they're already separate for external compasses so I think we can close this.  If it's a documentation issue then probably belongs on the wiki issues list (or maybe just go ahead and change it).

closed 2018-06-30 by rmackay9