ArduPilot/ardupilot #7865: PROPOSAL: make version releases for  APM HAL independently from top vehicles application

hiro2233, 2018-03-07:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

magicrub, 2018-03-07:
Issue details are blank. Please say something...

hiro2233, 2018-03-07:
@magicrub 👍

WickedShell, 2018-03-10:
I don't really get the issue. We don't release HAL builds, each vehicle's release is tagged across the HAL/features it needs. If we said AC 3.6 was based on HAL 1.0 isn't actually that interesting because nothing else will target that HAL anyways as it will have changed by the next vehicle release. Similarly none of the current features from the vehicle level would come close to compiling on the various older APM boards without massive changes and cutting anyways.

magicrub, 2018-03-11:
This is not in the scope of this autopilot project. Closing.

closed 2018-03-11 by magicrub