ArduPilot/ardupilot #7869: Plane: A bug in set_target_altitude_proportion() of path planning

luweikxy, 2018-03-08:
Hi, guys, maybe there is a small bug in set_target_altitude_proportion() when we are doing path planning.
Here is the website of this function:
https://github.com/ArduPilot/ardupilot/blob/1a3ca43e862e17c86057666bf07ccffd1c432eb7/ArduPlane/altitude.cpp#L266

when current altitude is 5 meters higher than target altitude, we do not want plane go down to the target altitude on the planned path, we want to plan another path to make the plane can fly to the target waypoint without going down, especially in auto takeoff. the above is part of purpose of this function.

The code in APM is:
~~~(C++)
void Plane::set_target_altitude_proportion(const Location &loc, float proportion)
{
    // ******omit several lines **************
    //
    if(g.glide_slope_threshold > 0) {
        if(target_altitude.offset_cm > 0 && calc_altitude_error_cm() < -100 * g.glide_slope_threshold) {
            set_target_altitude_location(loc);
            set_offset_altitude_location(loc);
            change_target_altitude(-target_altitude.offset_cm*proportion);
            //adjust the new target offset altitude to reflect that we are partially already done
            if(proportion > 0.0f)
                target_altitude.offset_cm = ((float)target_altitude.offset_cm)/proportion;
        }
    }
}
~~~

What I adjust is:
~~~(C++)
void Plane::set_target_altitude_proportion(const Location &loc, float proportion)
{
    // ******omit several lines **************
    //
    if(g.glide_slope_threshold > 0) {
        if(target_altitude.offset_cm > 0 && calc_altitude_error_cm() < -100 * g.glide_slope_threshold) {
            set_target_altitude_location(loc);
            set_offset_altitude_location(loc);
            //adjust the new target offset altitude to reflect that we are partially already done
            if(proportion > 0.0f)
                target_altitude.offset_cm = ((float)target_altitude.offset_cm)/proportion;
            change_target_altitude(-target_altitude.offset_cm*proportion);
        }
    }
}
~~~
[image]
But, if we donn't adjust the code order of this function, it will means that once the plane ii 5m higher than the target altitude, the target altitude will become "wrong_target_altitude", this is not our purpose. So we should adjust the code order as showed in picture.

After adjusted,  the target altitude will become "target_altitude_after_adjust" which is equal to current altitude. this is what we want.

magicrub, 2018-03-08:
PROTIP: in GitHub if you click on the line number it generates [a URL with the line number](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/altitude.cpp#L276).

[image]

Meanwhile, I believe @WickedShell worked on this. What do you think?

luweikxy, 2018-03-09:
@magicrub Thanks for your tip and sorry I am afraid I donn't know who can work on this.

magicrub, 2018-03-09:
I can make the change, that's the easy part. I'm just not totally convinced it's the right choice yet. I've pinged my wingman @wickedshell

WickedShell, 2018-03-10:
I'm aware of the issue opened here. I need to sit down this weekend and just scratch through some examples before responding. (It's been awhile since I've looked at this bit of the code, and it's much more complicated then it really needs to be)