ArduPilot/ardupilot #7872: Copter: follow mode should be able to use relative altitudes

rmackay9, 2018-03-08:
As mentioned in this related issue (https://github.com/ArduPilot/ardupilot/issues/7871), one issue we often see is that when a vehicle is switched to Follow mode it immediately climbs or descends a few meters because of differences in the estimated *absolute altitude* of the vehicle.  We can mostly remove this problem by allowing the user to select that *relative altitudes* be used.  In fact this should probably be the default because in most cases all vehicles are arming and taking off very close to each other so their relative altitudes should be very similar.

Steps to implement this change include:

- add a new FOLL_ALT_SOURCE parameter to the [AP_Follow library](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Follow/AP_Follow.cpp#L34).  "0" should be "absolute", "1" should be relative.
- in AP_Follow::handle_msg() function we fill in the _target_location structure.  We should check the value in the FOLL_ALT_SOURCE parameter and if it's "1" we should put the [GLOBAL_POSITION_INT](http://mavlink.org/messages/common#GLOBAL_POSITION_INT) message's relative_alt field into the _target_location.alt.  We should also set the _target_location.flags.relative_alt = 1.
- in [AP__Follow:get_target_dist_and_vel_ned we calculate the dist_vec](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Follow/AP_Follow.cpp#L173) using location_3d_diff_NED().  It seems this function does not check the Location structure's flags.relative_alt value so we can either change the current_loc value so it's an altitude above home (subtract the [AHRS.get_home().alt](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_AHRS/AP_AHRS.h#L471)) or a slightly more intrusive change would be to redefine _target_location to be a [Location_Class](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Common/Location.h) which handles these comparison better I think.

By the way, the absolute altitudes don't match because during startup the EKF/AHRS is initialising the ekf-origin/ahrs-home to a slight different altitude which is reported by each vehicle's GPS.

hoangthien94, 2018-06-03:
Implemented in this #8569 .  For the 3rd point I used the first approach (current_loc).

rmackay9, 2018-06-30:
This has been fixed by @hoangthien94 (thanks!) and will go out with Copter-3.6-rc3 in a few days.

closed 2018-06-30 by rmackay9