ArduPilot/ardupilot #7873: Maximum analog sampling rate?

padvinder95, 2018-03-08:
https://github.com/ArduPilot/ardupilot/blob/1baa6eb049ab7090b845ffe0b50c7978c1fb9e73/libraries/AP_HAL_PX4/AnalogIn.cpp#L257

Hi, what is the reason analog sampling is throttled to 100 Hz? Is it possible to go faster, e.g. 300 Hz?

I would like to sample an analogue sensor at fast loop speed (300–400 Hz) and tried to do that by changing the delta_t limit from 10000 µs (100 Hz) to 3333 µs (300 Hz). In my log, I see changed values only at 50, 100 and 150 Hz. (As though there is new data available only every 2 of these ticks at fastest—150 Hz—but in bad cases only after 6 ticks—50 Hz).

magicrub, 2018-03-08:
what hardware do you have?

padvinder95, 2018-03-08:
My bad. A pixhawk 2.4.8 or thereabouts.

IamPete1, 2021-07-26:
Looks like we have the same 100hz defualt rate on ChibiOS, 

https://github.com/ArduPilot/ardupilot/blob/f20644b6cf87b53c2394ad1ff07673c5ae657864/libraries/AP_HAL_ChibiOS/AnalogIn.cpp#L297

I see no reason you cant bump it up. I'm going to close this, please reopen if there is still a issue.

closed 2021-07-26 by IamPete1