ArduPilot/ardupilot #7875: PLANE: Aggressive pitch changes on flight mode switch

Naterater, 2018-03-08:
#### Issue details
When switching modes, changes in pitch are often dramatic. Aircraft aren't usually made for very aggressive pitch changes, and internal components are often dislodged when switching from a positive pitch mode like takeoff to FBWA at "0" pitch. Could a MODE_PITCH_SLEWRATE parameter be added to soften these otherwise aggressive pitch changes?

#### Version
3.8.4

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Traditional

#### Hardware type
Pixhawk 2.1

#### Logs
[image]

magicrub, 2018-03-08:
I suspect this only happens with the special case of changing mode MANUAL -> anything. Can you confirm it only happens when leaving MANUAL?

magicrub, 2018-03-08:
>switching from a positive pitch mode like takeoff to FBWA

do you mean AUTO -> FBWA? if so a pitch slew rate is not the solution, I suspect it's actually a control system problem of getting step inputs on the target airspeed or something like that. And then whatever that is would need to be slewed

Naterater, 2018-03-08:
> Can you confirm it only happens when leaving MANUAL?

I'll have to switch my transmitter to change mode without delay and without going to manual in-between switches. I'll report back. I was assuming it was a change in pitch demanded though that was causing the jerky behavior.

IamPete1, 2021-07-26:
Seems to have be caused by going via manual, I'm going to close, please re-open if there is a genuine issues.

closed 2021-07-26 by IamPete1