ArduPilot/ardupilot #7892: GPS-PPK Accuracy Improvement Needed

Naterater, 2018-03-10:
#### Issue details
Surveyors are using a standalone PPK system instead of ardupilot. All of the hardware is available -- the Here+ and a hotshoe feedback pin are already integrated, however ardupilot does not support the very low latency that is needed for sub-cm geotagging accuracy.

The problems that I am aware of
1.) CAM event timing at 50Hz
2.) GPS parsing delay (no PPS signal tracking)
3.) Cam offsets are not available/tracked
4.) GPS logs (converted from mission planner) do not contain event timestamps for RTKlib processing

Some more discussion about the issue: https://discuss.ardupilot.org/t/here-ppk-and-hotshoe-feedback-processing/26666/8

#### Version
All

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
AIrframes moving during image capture

#### Hardware type
Pixhawk, Here+, Hotshoe feedback cable

#### Logs
N/A

Saijin-Naib, 2018-11-12:
This would make my surveys orders of magnitude more robust, and would really make my Solo an even more powerful value proposition.

rmackay9, 2018-11-13:
Discussed on the dev call and we think that @bugobliterator is looking into this.

moreba1, 2025-01-24:
Hello , This is good work , 

1- Also "uBlox dual-band RAW logging for PPK" must be implemented for F9P GPS . https://github.com/ArduPilot/ardupilot/pull/25941

@Naterater  2- Does it possible that .obs and .nav file  (or .UBX file ) saved different from of .BIN log file ?

3-  I have read that there is need to convert .bin file to Rinex file by Mission Planner.  Often log file of total flight has large size and downloading it takes a lot time .

4- 50 HZ for CAM Event timing is not enough. for a 20 m/s velocity fixed wing drone and 1 cm accuracy there is need 2000 HZ time mark accuracy.