ArduPilot/ardupilot #7897: getting errors while running SITL on windows

OmarShadeed, 2018-03-11:
hi, 
Ive followed al the steps listed here http://ardupilot.org/dev/docs/sitl-native-on-windows.html to run SITL using arducopter but whenever i try    " make sitl -j4 "  i get this
$ make sitl -j4
../mk/apm.mk:25: /modules/uavcan/libuavcan/include.mk: No such file or directory
make: *** No rule to make target '/modules/uavcan/libuavcan/include.mk'.  Stop.

and whenever i try any of these commands to start the SITL  
"sim_vehicle.py -j4 -L KSFO"    or    " sim_vehicle.py -j4 --map --console"

this is what i get 

$ sim_vehicle.py -j4 --map --console
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at -35.363261,149.165230,584,353 (CMAC)
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/Omar Shadeed/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Setting top to                           : /home/Omar Shadeed/ardupilot
Setting out to                           : /home/Omar Shadeed/ardupilot/build
Autoconfiguration                        : enabled
Setting board to                         : sitl
Checking for 'g++' (C++ compiler)        : /usr/bin/g++
Checking for 'gcc' (C compiler)          : /usr/bin/gcc
Checking for need to link with librt     : not necessary
Checking for HAVE_CMATH_ISFINITE         : yes
Checking for HAVE_CMATH_ISINF            : yes
Checking for HAVE_CMATH_ISNAN            : yes
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes
Checking for header endian.h                   : yes
Checking for header byteswap.h                 : yes
Checking for program 'python'                  : /usr/bin/python
Checking for python version >= 2.7.0           : 2.7.14
Checking for program 'python'                  : /usr/bin/python
Checking for python version >= 2.7.0           : 2.7.14
Source is git repository                       : yes
Update submodules                              : yes
Checking for program 'git'                     : /usr/bin/git
Checking for program 'size'                    : /usr/bin/size
Benchmarks                                     : disabled
Unit tests                                     : enabled
Checking for program 'rsync'                   : /usr/bin/rsync
'configure' finished successfully (5.630s)
SIM_VEHICLE: Building
SIM_VEHICLE: "/home/Omar Shadeed/ardupilot/modules/waf/waf-light" "build" "--target" "bin/arducopter" "-j" "4"
Waf: Entering directory `/home/Omar Shadeed/ardupilot/build/sitl'
Waf: Leaving directory `/home/Omar Shadeed/ardupilot/build/sitl'

BUILD SUMMARY
Build directory: /home/Omar Shadeed/ardupilot/build/sitl
Target          Text     Data   BSS  Total
--------------------------------------------
bin/arducopter  1840117  52716  416  1893249

Build commands will be stored in build/sitl/compile_commands.json
'build' finished successfully (8.627s)
SIM_VEHICLE: Using defaults from (/home/Omar Shadeed/ardupilot/Tools/autotest/default_params/copter.parm)
SIM_VEHICLE: Run ArduCopter
SIM_VEHICLE: "/home/Omar Shadeed/ardupilot/Tools/autotest/run_in_terminal_window.sh" "ArduCopter" "/home/Omar Shadeed/ardupilot/build/sitl/bin/arducopter" "-S" "-I0" "--home" "-35.363261,149.165230,584,353" "--model" "+" "--speedup" "1" "--defaults" "/home/Omar Shadeed/ardupilot/Tools/autotest/default_params/copter.parm"
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "/usr/bin/cygstart" "-w" "/cygdrive/c/Program Files (x86)/MAVProxy/mavproxy.exe" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501" "--out" "127.0.0.1:14550" "--out" "127.0.0.1:14551" "--map" "--console"
RiTW: Starting ArduCopter : /home/Omar Shadeed/ardupilot/build/sitl/bin/arducopter -S -I0 --home -35.363261,149.165230,584,353 --model + --speedup 1 --defaults /home/Omar Shadeed/ardupilot/Tools/autotest/default_params/copter.parm
which: no osascript in (/usr/local/bin:/usr/bin:/cygdrive/c/ProgramData/Oracle/Java/javapath:/cygdrive/c/Program Files (x86)/Intel/iCLS Client:/cygdrive/c/Program Files/Intel/iCLS Client:/cygdrive/c/Windows/system32:/cygdrive/c/Windows:/cygdrive/c/Windows/System32/Wbem:/cygdrive/c/Windows/System32/WindowsPowerShell/v1.0:/cygdrive/c/Program Files/MATLAB/R2016b/runtime/win64:/cygdrive/c/Program Files/MATLAB/R2016b/bin:/cygdrive/c/Program Files/MATLAB/R2016b/polyspace/bin:/cygdrive/c/Program Files (x86)/MAVProxy:/cygdrive/c/Users/Omar Shadeed/AppData/Local/Microsoft/WindowsApps:/cygdrive/c/Users/Omar Shadeed/AppData/Local/GitHubDesktop/bin:/home/Omar Shadeed/ardupilot/Tools/autotest)
which: no xterm in (/usr/local/bin:/usr/bin:/cygdrive/c/ProgramData/Oracle/Java/javapath:/cygdrive/c/Program Files (x86)/Intel/iCLS Client:/cygdrive/c/Program Files/Intel/iCLS Client:/cygdrive/c/Windows/system32:/cygdrive/c/Windows:/cygdrive/c/Windows/System32/Wbem:/cygdrive/c/Windows/System32/WindowsPowerShell/v1.0:/cygdrive/c/Program Files/MATLAB/R2016b/runtime/win64:/cygdrive/c/Program Files/MATLAB/R2016b/bin:/cygdrive/c/Program Files/MATLAB/R2016b/polyspace/bin:/cygdrive/c/Program Files (x86)/MAVProxy:/cygdrive/c/Users/Omar Shadeed/AppData/Local/Microsoft/WindowsApps:/cygdrive/c/Users/Omar Shadeed/AppData/Local/GitHubDesktop/bin:/home/Omar Shadeed/ardupilot/Tools/autotest)
which: no konsole in (/usr/local/bin:/usr/bin:/cygdrive/c/ProgramData/Oracle/Java/javapath:/cygdrive/c/Program Files (x86)/Intel/iCLS Client:/cygdrive/c/Program Files/Intel/iCLS Client:/cygdrive/c/Windows/system32:/cygdrive/c/Windows:/cygdrive/c/Windows/System32/Wbem:/cygdrive/c/Windows/System32/WindowsPowerShell/v1.0:/cygdrive/c/Program Files/MATLAB/R2016b/runtime/win64:/cygdrive/c/Program Files/MATLAB/R2016b/bin:/cygdrive/c/Program Files/MATLAB/R2016b/polyspace/bin:/cygdrive/c/Program Files (x86)/MAVProxy:/cygdrive/c/Users/Omar Shadeed/AppData/Local/Microsoft/WindowsApps:/cygdrive/c/Users/Omar Shadeed/AppData/Local/GitHubDesktop/bin:/home/Omar Shadeed/ardupilot/Tools/autotest)
which: no gnome-terminal in (/usr/local/bin:/usr/bin:/cygdrive/c/ProgramData/Oracle/Java/javapath:/cygdrive/c/Program Files (x86)/Intel/iCLS Client:/cygdrive/c/Program Files/Intel/iCLS Client:/cygdrive/c/Windows/system32:/cygdrive/c/Windows:/cygdrive/c/Windows/System32/Wbem:/cygdrive/c/Windows/System32/WindowsPowerShell/v1.0:/cygdrive/c/Program Files/MATLAB/R2016b/runtime/win64:/cygdrive/c/Program Files/MATLAB/R2016b/bin:/cygdrive/c/Program Files/MATLAB/R2016b/polyspace/bin:/cygdrive/c/Program Files (x86)/MAVProxy:/cygdrive/c/Users/Omar Shadeed/AppData/Local/Microsoft/WindowsApps:/cygdrive/c/Users/Omar Shadeed/AppData/Local/GitHubDesktop/bin:/home/Omar Shadeed/ardupilot/Tools/autotest)
RiTW: Window access not found, logging to /tmp/ArduCopter.log
SIM_VEHICLE: MAVProxy exited
SIM_VEHICLE: Killing tasks
 
if anyone could help i'd be thankful.

rmackay9, 2018-03-11:
Hi Omar,

Thanks for the report.  We have recently been improving the developer wiki but haven't gotten to this particular page yet.  This somewhat similar one is done though: http://ardupilot.org/dev/docs/building-setup-windows-cygwin.html

In any case, you can skip the "make" step because that will be done by the sim_vehicle call in any case.

It looks like the issue is that it can't pop up a terminal.  I suspect if the "--map --console" is left out of the call to sim_vehicle then it will at least startup the simulator.  You won't be able to see anything though so it's just a test.

I think we will need to go through that page and see what's missing from the instructions - probably the installation of a terminal program.

peterbarker, 2018-03-12:
On Sun, 11 Mar 2018, OmarShadeed wrote:

> make: *** No rule to make target '/modules/uavcan/libuavcan/include.mk'.
> Stop.
>

Try `git submodule update --init --recursive`

OmarShadeed, 2018-03-12:
@peterbarker thanks for ur reply. 
 I tried  `git submodule update --init --recursive` and then ran " sim_vehicle.py -j4 --map --console" again  but still i get the same problem.

OmarShadeed, 2018-03-12:
@rmackay9 thanks a lot for ur reply. 
I've tried the command 'sim_vehicle.py -j4` without  "--map --console" but i still get the same error. 
I followed the instructions in http://ardupilot.org/dev/docs/building-setup-windows-cygwin.html it has same instructions i followed while installing Cygwin so nothing changed. I even uninstalled Cygwin and reinstalled cause i thought maybe one of the packages are missing or something but i still get the same error.

peterbarker, 2018-03-13:
@OmarShadeed Please provide the content of `/tmp/ArduCopter.log`

The long error message there can probably be ignored.  It's a MacOSX thing.

OmarShadeed, 2018-03-13:
@peterbarker I tried '/tmp/ArduCopter.log' and this is what it gave me 

Omar Shadeed@DESKTOP-2NHP0QJ ~/ardupilot/ArduCopter
$ /tmp/ArduCopter.log
/tmp/ArduCopter.log: line 1: /home/Omar: No such file or directory
 I dunno why it's still not working.

OmarShadeed, 2018-03-13:
@peterbarker I'm running a windows 10 boot camp on my mac. Does that might be the problem ?

peterbarker, 2018-03-14:
On Tue, 13 Mar 2018, OmarShadeed wrote:

> @peterbarker I'm running a windows 10 boot camp on my mac. Does that might
> be the problem ?

Nope.  The problem is the space in your username....

peterbarker, 2018-03-14:
On Tue, 13 Mar 2018, OmarShadeed wrote:

> @peterbarker I tried '/tmp/ArduCopter.log' and this is what it gave me
> 
> Omar Shadeed@DESKTOP-2NHP0QJ ~/ardupilot/ArduCopter
> $ /tmp/ArduCopter.log
> /tmp/ArduCopter.log: line 1: /home/Omar: No such file or directory
> I dunno why it's still not working.

Sorry, the problem is *almost certainly* the space in your username.

Please paste the contents of that file in.  You can see the content using:
cat /tmp/ArduCopter.log

OmarShadeed, 2018-03-14:
@peterbarker  This is what i get when i run  'cat /tmp/ArduCopter.log'

$ cat /tmp/ArduCopter.log
/home/Omar Shadeed/ardupilot/Tools/autotest/run_in_terminal_window.sh: line 53: /home/Omar: No such file or directory

I still dont get what's the problem.

peterbarker, 2018-03-14:
On Wed, 14 Mar 2018, OmarShadeed wrote:

> @peterbarker This is what i get when i run 'cat /tmp/ArduCopter.log'
> 
> $ cat /tmp/ArduCopter.log
> /home/Omar Shadeed/ardupilot/Tools/autotest/run_in_terminal_window.sh: line 53: /home/Omar: No such file or directory
> 
> I still dont get what's the problem.

Space is used to separate arguments on a Unix command-line.  You're in for 
a world of pain if you include a space in your username!

I believe I've spotted the bug in this script which makes it an issue 
here.  If I create a pull request, are you able to test it for me?

OmarShadeed, 2018-03-14:
@peterbarker sure i can test it for u

OmarShadeed, 2018-03-14:
@peterbarker 
I tried to remove the space in my user name but i couldnt. Do I have to create a new account ?

peterbarker, 2018-03-14:
On Wed, 14 Mar 2018, OmarShadeed wrote:

> @peterbarker sure i can test it for u

https://github.com/ArduPilot/ardupilot/pull/7920

peterbarker, 2018-03-14:
On Wed, 14 Mar 2018, OmarShadeed wrote:

> @peterbarker
> I tried to remove the space in my user name but i couldnt. Do I have to create a new account ?

You don't, but it's a non-trivial exercise to do so.

If you're not heavily invested in your current account, create a new one 
:-)

OmarShadeed, 2018-03-14:
@peterbarker I renamed my account username and now in the users it appears without any spaces and i deleted Cygwin64 and reinstalled it but when i open it for the first time it still get my old username with space in it. Now i really dont know what to do.

peterbarker, 2018-03-14:
On Wed, 14 Mar 2018, OmarShadeed wrote:

> @peterbarker I renamed my account username and now in the users it appears without any spaces and i deleted Cygwin64 and reinstalled it but when i open it
> for the first time it still get my old username with space in it. Now i really dont know what to do.

Test my PR?

Sorry, I've no idea on why it's retained your old username there.  I don't 
use Windows or CygWin day-to-day.

OmarShadeed, 2018-03-19:
@peterbarker I've just tried it and it's finally working.
Thank you so much for your time. 
I just have one last question, does this fix the problem forever I mean I won't get any errors related to this problem again right ?

peterbarker, 2018-03-19:
On Mon, 19 Mar 2018, OmarShadeed wrote:

> @peterbarker I've just tried it and it's finally working.

Excellent, thanks!.  That code's in master, so you should be good to run 
that.

> I just have one last question, does this fix the problem forever I mean I
> won't get any errors related to this problem again right ?

That should be the case, yes.

Ilia-SB, 2018-08-31:
Just in case this helps. I also followed the steps to run SITL natively on Windows 10 and I wasgetting the same error messages when running sim_vehicle.py, namely:
no osascript in...
no xterm in...
no konsole in...
no gnome-terminal in...
though Mavproxy, Console and Map windows appear and everything seems to be running OK. Resolved this by re-running Cygwin install and selecting packages "xterm", "konsole" and "gnome-terminal". This helped with the 3 respective errors. As for the osascript, I managed to install it with "pip2 install osascript", but am still getting the error message even after adding the osascript folder to my PATH.
Anyway, as I've said, everything seems to be working just fine.

closed 2018-10-15 by EShamaev

priyamshah112, 2018-12-28:
> pip2 install osascript

I got similar errors thanks to your reply for solving 3 errors but can you help me out with where have you placed the mavproxy.exe file in your cygwin64? as mavproxy, Console and Map windows do not appear the errors shown are
RiTW: Window access not found, logging to /tmp/ArduSub.log
Error loading Python DLL 'C:\cygwin64\bin\python27.dll'.
LoadLibrary: The specified module could not be found.

Ilia-SB, 2019-01-09:
Sorry, can't help you here. I found out that setting up an Ubuntu virtual machine was easier for development and testing.