ArduPilot/ardupilot #7899: Rover: enhance crash check with angle limit

rmackay9, 2018-03-11:
We should improve Rover's crash check so that it changes the vehicle to Hold if it goes beyond a certain angular limit for a certain period of time.

The existing crash check is in the [crash_check.cpp](https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/crash_check.cpp) file but it' purpose is to check if the vehicle gets stuck on something in Auto, Guided, RTL or Smart RTL.

A new parameter should be added ([see this guide](http://ardupilot.org/dev/docs/code-overview-adding-a-new-parameter.html)), perhaps called "CRASH_ANGLE" which will hold the lean angle (in degrees).  It's possible that we may want to add a "CRASH_TIMEOUT" parameter as well to hold a time (in seconds) which the high lean angle must persist for the crash check to trigger.

The crash_check() method should have two parts to it.  One part will do what it does now, check if the vehicle gets stuck in autonomous modes.  The 2nd part should:

- calculate the vehicle's current lean angle.  This could be something like "float lean_angle = degrees(safe_sqrt(sq(ahrs.roll)+sq(ahrs.pitch)))".
- check if the lean-angle is over the new CRASH_ANGLE parameter value, if yes, increment the counter.

Because there are now two checks, the code will need to reset the counter back to zero if neither of the two checks triggers.

mikerob, 2018-03-12:
Would it be sensible to include a threshold cross-track error in the "crash" check?  If the vehicle is deviating from the planned path more than a user-definable amount, it has potentially suffered some sort of actuator or mechanical failure.  You'd need to be fairly generous since most vehicles deviate a fair bit at waypoint corners, but more than 2-3 metres would indicate a problem.

khancyr, 2018-03-12:
linked to https://github.com/ArduPilot/ardupilot/issues/6107, but this one is better described

sarthak268, 2018-04-29:
is this still open? If yes, please assign.

peterbarker, 2018-04-30:
@sarthak268 For various reasons I can't assign this to you.  I've assigned this to myself on the assumption that you're going to work on it - let me know if you get stuck, and have fun!

sarthak268, 2018-04-30:
the second part of the code is to be added to before the part where we are incrementing the crash counter(crash_counter++) and after the part before it?

rmackay9, 2018-07-06:
The angle based crash check is in master now - it was included as part of the balance bot enhancements.

closed 2018-07-06 by rmackay9